RobotLocomotion/drake #2979: Move RigidBodyForceElement and RigidBodySensor Vectors into RigidBodyTree

liangfok, 2016-07-29:
As requested by @amcastro-tri here: https://reviewable.io/reviews/RobotLocomotion/drake/2976#-KNqrkEnixWRftsq4j3t

The motivation is `RigidBodyForceElement` and `RigidBodySensor` are not "systems" in and of themselves. Instead, they are more like things that are contained in and attached to a `RigidBodyTree`.

liangfok, 2016-08-15:
@amcastro-tri: This may be easier said than done because doing this will change the composition of `RigidBodyTree`'s state vector, which is currently `[joint position values][joint velocity values]`. If we decide to move sensors and actuators into the `RigidBodyTree`, its state vector may need to be `[joint position values][joint velocity values][sensor state values][actuator state values]`. Significant code changes will be required in both `RigidBodyTree` and `RigidBodySystem`.

liangfok, 2017-04-12:
@amcastro-tri, I'm going to close this issue since I assume it'll be superseded by the switch to `MultibodyWorld`. Please re-open if I'm wrong.

closed 2017-04-12 by liangfok

amcastro-tri, 2017-04-12:
Thank you @liangfok. Yes, MBT will provide `ForceElement` objects.