RobotLocomotion/drake #2982: git clone of spdlog failed

sosuiming, 2016-07-29:
output of make:

Cloning into 'externals/spdlog'...
Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
Clone of 'git@github.com:gabime/spdlog.git' into submodule path 'externals/spdlog' failed
CMakeFiles/spdlog.dir/build.make:88: recipe for target 'spdlog-prefix/src/spdlog-stamp/spdlog-download' failed
make[2]: *** [spdlog-prefix/src/spdlog-stamp/spdlog-download] Error 1
CMakeFiles/Makefile2:366: recipe for target 'CMakeFiles/spdlog.dir/all' failed
make[1]: *** [CMakeFiles/spdlog.dir/all] Error 2
Makefile:94: recipe for target 'all' failed
make: *** [all] Error 2

liangfok, 2016-07-29:
It looks like you need to upload your public SSH key to your github account:

https://help.github.com/articles/adding-a-new-ssh-key-to-your-github-account/

liangfok, 2016-07-29:
Actually this PR should have fixed the problem:

https://github.com/RobotLocomotion/drake/pull/2977

Do you have it in your local clone?

sosuiming, 2016-07-29:
I changed the url in the hidden file: .gitmodules to:
url = https://github.com/gabime/spdlog.git
and the installation proceeds properly.
Thanks very much.
I think this is better mentioned in the faq.

liangfok, 2016-07-29:
I'm glad you got it to work!

closed 2016-07-29 by liangfok