RobotLocomotion/drake #2985: Implement Gain system

amcastro-tri, 2016-07-29:
For this system:
- It is direct feedthrough
- `y = k * u` where k is the gain constant.

Where should the constant of the system go? somehow into the context as a "parameter" so we can allow four auto-differentiation? Thoughts @david-german-tri, @sherm1?

sherm1, 2016-07-29:
> Where should the constant of the system go? 

Gain should be a Parameter to the system block so that it can be exposed at the System Diagram level like an Input. We don't have that yet though so I would suggest just making it a type `T` member of the `Gain<T>` System class. We can still differentiate with respect to it but to do so you have to know there is a Gain block inside and dig it out to get to its Gain-specific methods.

david-german-tri, 2016-07-29:
Agreed with @sherm1 on all counts - the expedient thing to do right now is just pass it into the constructor.

amcastro-tri, 2016-08-22:
Closed by #3097

closed 2016-08-22 by amcastro-tri