RobotLocomotion/drake #2986: Basic source systems

amcastro-tri, 2016-07-29:
Implement systems with no input nor state that output a given signal. Basic examples to implement are:
- ConstantSource: `y(t) = k`
- HarmonicSource: `y(t) = a*sin(w*t+phi)`

sherm1, 2016-07-29:
One issue for `ConstantSource`: it doesn't actually have to produce a vector output, it can serve up any abstract value. In the System3 example, I made a VectorConstantSystem so I could have one that would promise to provide a vector. Since System 2 doesn't yet support abstract values, you might want just to produce VectorConstantSource for now.

Also, the constant itself should (eventually) be a Parameter of the ConstantSource so that we can conveniently mess around with it, such as autodiff it.

HarmonicSource would be limited to vector output I think, with vector Parameters for a,w,phi.

sherm1, 2016-07-29:
Simulink is always a good source for inspiration. Here are its [sources](http://www.mathworks.com/help/simulink/sources.html), and its general [block library](http://www.mathworks.com/help/simulink/block-libraries.html).

closed 2017-02-01 by amcastro-tri