RobotLocomotion/drake #2987: Implement differentiator system

amcastro-tri, 2016-07-29:
This system outputs the derivative of the input signal.
In order to do so this system will need to (at least) save the previous time step value of its input. 
This previous time step value will be saved as a "mode" in the context, do you agree @david-german-tri and @sherm1?

Differentiator systems should be avoided whenever possible, [see Mathworks article here.](http://www.mathworks.com/help/simulink/ug/best-form-mathematical-models.html#bra6aje)

sherm1, 2016-07-29:
> This previous time step value will be saved as a "mode" in the context, do you agree @david-german-tri and @sherm1?

This could *almost* be an ordinary difference variable with update `xd[n]=u(t_n)` (where u is the continuous input we want to differentiate and t_n the time at the start of the n'th time step). Then the output is `y(t_i)=(u(t_i)-xd[i-1])/dt`. The problem is the `dt`. We need somewhere to record the sample time at which xd was recorded. So we could use a mode variable to hold both the value and update time. Probably it would be better always to record the previous step time separately since sample-every-step is a common sample time specification. If we did that, then an ordinary difference variable could be used here.

That said, that is not a great numerical method for calculating derivatives. Simulink's [Derivative block](http://www.mathworks.com/help/simulink/slref/derivative.html) includes some optional filtering that may help produce a better result. Its [PID block](http://www.mathworks.com/help/simulink/slref/pidcontroller.html) also has a filter involved in its derivative calculator.

tri-ltyyu, 2019-03-22:
Use case likely outdated. Closing for now; please re-open if needed.
Discrete derivative is currently available.

closed 2019-03-22 by tri-ltyyu