RobotLocomotion/drake #2989: Implement delay system.

amcastro-tri, 2016-07-29:
The output of the system is its input `tau` seconds in the past:
- y(t) = u(t-tau)
- Direct feed through: no.

This is an interesting system because it needs to implement a buffer to save previous instances of the input `u`. This is not a state (no xdot) neither a difference variable (no Delta equation) and therefore falls in our category of "mode" variables.
Thoughts @david-german-tri and @sherm1?

sherm1, 2016-07-29:
Yes, the buffer is a discrete variable of abstract type so would be include in our catch-all "mode" variable category. It's "sample time" would be what Simulink calls "every major step", that is, it gets updated every time the system trajectory is irreversibly advanced.

The direct feed-through issue is tricky for delay buffers because of the possibility that the previous time step is longer than the requested time delay. In that case the delayed value must be calculated by *extrapolation* rather than *interpolation* and extrapolation is notoriously bad numerically. If direct feedthrough is permitted then the current value of the signal can be used so that interpolation is once again possible. That is discussed in the doxygen for Simbody's Delay buffer [here](https://simbody.github.io/simbody-master-doxygen/api/classSimTK_1_1Measure___1_1Delay.html#details). Direct feedthrough is not necessarily a bad thing; it is only a problem when the output is fed back to the input to form a loop.

david-german-tri, 2016-07-29:
To support this, I suspect that adding `mode_state` to the State struct, and some new `System` function to update it, is a pretty small change.  You probably want to wait for #2872, though.

david-german-tri, 2016-09-29:
#3587 is nearly here!

naveenoid, 2016-09-29:
I can take a pass at this issue but I need some input from @david-german-tri

amcastro-tri, 2016-09-29:
Ok, feel free to un-assign me and assign yourself to this @naveenoid. Please ping me in that PR since I'm interested to see how that works.

sherm1, 2016-09-29:
I have implemented these before if you want to discuss. Simbody's Delay documentation mentions some of the ugly issues [here](https://simtk.org/api_docs/simbody/latest/classSimTK_1_1Measure___1_1Delay.html#details). Simulink's discussion for its TransportDelay block is [here](https://www.mathworks.com/help/simulink/slref/transportdelay.html). These are quite tricky to implement correctly.

tri-ltyyu, 2019-03-22:
Discrete derivative now avail. Please re-open if still needed

closed 2019-03-22 by tri-ltyyu