RobotLocomotion/drake #3005: Problem with Bertini External

liangfok, 2016-07-31:
# The error

```
$ cd drake-distro
$ git submodule update
Cloning into 'externals/bertini'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:RobotLocomotion/bertini.git' into submodule path 'externals/bertini' failed
```

jamiesnape, 2016-08-01:
It exists and is private, so this behavior is by design and not a bug.

liangfok, 2016-08-01:
Ah. That explains why I did not see it. Thanks.

closed 2016-08-01 by liangfok

soonho-tri, 2016-08-08:
@liangfok, I have the same problem. Do you think an easy solution is to ask a permission for `RobotLocomotion/bertini` repository?

@jamiesnape, I added a submodule on machine A and made a commit. I moved to machine B, pulled the commit. Now, I want to update submodules, so I do `git submodule update --recursive` but this issue terminates the update process in the middle. Is there an easy workaround (other than having a read permission on bertini repo)?

jamiesnape, 2016-08-08:
CMake will run the submodule update for you automatically. If you really do not want to do that, update the submodules independently instead of using recursive.

soonho-tri, 2016-08-08:
@jamiesnape, I see. Thank you!