RobotLocomotion/drake #3017: store the trajectory result in simulation

hongkai-dai, 2016-08-01:
Currently `Simulation.h->simulate` functions only outputs the terminal time of the simulation. It will be useful to output the simulated states along the trajectory.

There are two approaches to do this

1. Revise the return argument of the simulate, to output two vectors: A vector `vector<double> t_simulate` and  `vector<OutputVector> y_simulate` directly. These vectors are empty at the beginning of the simulation, and then append the time and the output at i'th time sample at every step. We can also create a struct `SimulationResult` to store these two vectors.
2. cascade a logging system with the simulated system, as `simulated_system->logging_system`, and the logging system should store the trajectory. I do not know a very good way to implement this, since it would require that the dimension of the logging system increases at every simulation time step, to append the simulation result at that time step.

jwnimmer-tri, 2016-08-01:
/CC @david-german-tri @sherm1 

I think this looks much different in System 2.0 (much more simulink-like) than System 1.0.  We should get these two folks thoughts.

naveenoid, 2016-08-01:
I am personally definitely for option 2 but then again, perhaps its already being considered in System2.0 design?

sherm1, 2016-08-01:
(had a brief f2f with @hongkai-dai) 
I think "cascading" a trajectory-saving system (option 2) is the best approach. Other than syntactic details, this is just what we would do in System 2 also.

The trajectory-logger System could be as simple as an `std::vector<T>` where T is the type of its input vector (same as the output vector Hongkai wants to save). Each time one arrives, `push_back` it onto the `std::vector`. Then after the simulation is done you'll have the whole trajectory available for further analysis (through some custom methods of the logger System).

liangfok, 2016-08-01:
Memory limits may make storing the entire history in memory all at once. If this is the case, the `TrajectoryLogger` system could simply publish the state on an LCM channel or ROS topic and allow another process to save the state. Another option is to store the state in a plain text file using the newly added `spdlog`-based logging system (https://github.com/RobotLocomotion/drake/pull/2954). Of course, all of these options increase the probability of message loss. Message losses can be detected through sequence numbers.

hongkai-dai, 2016-08-01:
@liangfok , publishing to LCM channel and ROS topic would be great. It can be an option of the `TrajectoryLogger` class to support these features. We can start with the `vector` idea as @sherm1  mentioned, in that way we can test the system without any external dependency.

david-german-tri, 2016-08-01:
I agree with all of the above.  A logging system that you cascade in is the right approach.

In Q4 or Q1 I hope to provide a generalized System2 infrastructure for logging to network or disk. In the meantime, an application-specific system seems totally reasonable, for System1 or System2.

hongkai-dai, 2016-08-01:
FWIW, ideally this trajectory logging system should log a trajectory (like a piece-wise polynomial trajectory), instead of just the sample values at the trajectory at every `dt`. By using different integration schemes (Euler versus higher order Runge-Kutta), two trajectories with the same value at the sample time may correspond to different values in between the time sample samples. 

As the first pass, for simplicity, I will not interpolate the samples to generate the trajectory.

PR #3051 (2016-08-09): add a trajectory logger
Resolves #3017

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3051)
closed 2016-08-09 by hongkai-dai