RobotLocomotion/drake #3023: Update Documentation Generation Instructions To Not Distinguish Between In-Source vs. Out-of-source

liangfok, 2016-08-01:
This is because we now only support on form of build that is in `drake-distro/build/`.

The file to update is: `drake-distro/drake/doc/documentation_instructions.rst`

sherm1, 2016-08-01:
> This is because we now only support on form of build that is in drake-distro/build/

I don't think that's correct, at least not on Windows. Why wouldn't we support an out-of-source build, which is the generally-recommended strategy when using CMake?

jwnimmer-tri, 2016-08-01:
FWIW, I was skeptical of `drake-distro/build` as the build directly, but I've been loving it since I started using it.  It means that `git` operations work even if your cwd is in the build directory, that `rm -rf build` is still just as easy as with any other directory setup, and that it's easy to have multiple source checkouts should you want such a thing.

david-german-tri, 2016-08-01:
> Why wouldn't we support an out-of-source build, which is the generally-recommended strategy when using CMake?

I also had this question.  Here's my best attempt at a short answer:

* We have to document/support a specific build directory for three reasons:
 1. MATLAB code has to execute in-source, but also needs to locate the build directory.  (See `addpath_drake`)
 2. Without a concrete build directory, lots of instructions become hard to read.
 3. There needs to be some default build directory that CI uses.
* We use `drake-distro/build` in particular because it was Russ's preference.

Like @jwnimmer-tri, I started out skeptical, and have warmed to neutral over the past few weeks.

sherm1, 2016-08-01:
Don't you need multiple build directories for Debug & Release builds?

david-german-tri, 2016-08-01:
Multiple build directories are a great example.  They mostly work, except for some externals that aren't fully out of source.  If developers opt to create Debug and Release build directories, they will generally have no problems.

However, that setup is not documented or tested in CI.  PRs that break it are not _ipso facto_ bugs.  The build cop will not roll them back.  There are known sharp edges.  Hence, it's "not supported", in the sense I believe @liangfok is using here.

sherm1, 2016-08-01:
I like the idea of having a standard build location that is well tested and used consistently in documentation. But I hope the real out-of-source builds will continue to work -- my Windows workflow is very different than the Linux one (basically zero use of command line for anything), and multiple out-of-source build directories is typical for me.

liangfok, 2016-08-02:
Yeah, by "not supported" I simply mean it's not officially documented or covered by CI. It's definitely possible that out-of-source builds still work. In my bespoke setup on Linux and OS X, I actually have numerous clones of Drake each focusing on a different PR / issue and/or build type and/or build tool. We're going to need to update the Windows instructions: #3026.

jwnimmer-tri, 2016-08-02:
@sherm1
This term "out of source" -- I do not think it means what you think it means.  (Or at least, we disagree).

Consider three of the possible ways a build might exist
- (1) source code is in `root/dir1/foo.cc` and `root/dir2/bar.cc`, object code is in `root/dir1/foo.o` and `root/dir2/bar.o`
- (2) source code is in `root/dir1/foo.cc` and `root/dir2/bar.cc`, object code is in `root/build/dir1/foo.o` and `root/build/dir2/bar.o`
- (3) source code is in `root/dir1/foo.cc` and `root/dir2/bar.cc`, object code is in `other/dir1/foo.o` and `other/dir2/bar.o`

Option (1) is clearly awful and nobody should ever do that.  This is what I would call an "in-source" build.

I argue that (2) and (3) are equally useful for the cases you mention, but it seems like you are arguing that (3) is more useful than (2).

In both cases, I can nuke a single directory to reset all state -- in (2) it is `root/build/`; in (3) it is `other`.  In both cases, I can create multiple build-variants from the same source tree -- in (2) it is `root/build/debug` and `root/build/release`; in (3) it is `other` and `another`.

The benefits of (2) are as given in the upthread comments by David; the i, ii, iii.  I cannot see any additional advantage of (3) that would merit sacrificing portions of i,ii,iii.

sherm1, 2016-08-02:
Thanks for the explanations, @jwnimmer-tri and @david-german-tri. 

By "out of source" I mean, "not in my git repo". Since `drake-distro` is the root of the repo, I consider `drake-distro/build` to be in source. It is admittedly somewhat fuzzy with Drake since all the meaningful source is in `drake-distro/drake` and `drake-distro/build` is parallel to that. Also it appear that a build at the superbuild level writes into `drake-distro/externals` regardless of where the binaries are sent. 

I didn't realize that `drake-distro/build/debug` and `build/release` would be OK in this scheme. Does that still retain David's "i" benefit for Matlab to find the build files? I was assuming that everything has to go right below `build` so that `build/drake` would exist.

david-german-tri, 2016-08-02:
> Does that still retain David's "i" benefit for Matlab to find the build files? I was assuming that everything has to go right below build so that build/drake would exist.

Not at the moment, but I think it would be totally reasonable to add some more paths [here](https://github.com/RobotLocomotion/drake/blob/master/drake/addpath_drake.m#L129).

jwnimmer-tri, 2016-08-02:
> Also it appear that a build at the superbuild level writes into drake-distro/externals regardless of where the binaries are sent.

This is a defect, in the process of being fixed, and of course affects all choices of build directory location equally.

PR #4656 (2017-01-03): Freshen doc generation instructions.
@liangfok for feature review, @sammy-tri for platform review.  Fixes #3023.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4656)
closed 2017-01-03 by david-german-tri