RobotLocomotion/drake #3024: OS X instructions homebrew.rst are stasle

jwnimmer-tri, 2016-08-01:
https://github.com/RobotLocomotion/drake/blob/9c501d8ea16e7e90c0885b60b69e1f9853b7fd8d/drake/doc/homebrew.rst

The instructions say:
```
cd drake-distro
make options  # ...
make download-all
```

TTBOMK, none of that is needed, and simply continuing with `from_source.rst` is good enough -- it has instructions for the cmake and build steps.

For this issue, we should both fix the docs, and review the related PRs to `ubuntu.rst` or whatever to see why `homebrew.rst` got overlooked, and if anywhere else we still have lies.

jwnimmer-tri, 2016-08-01:
I am hoping somebody with OS X can pick this one up and fix / test it.

jamiesnape, 2016-08-02:
Maybe also rename to `mac.rst` now there is no MacPorts.

jwnimmer-tri, 2017-01-17:
I think this is done now?

soonho-tri, 2017-01-17:
The current documentation is good.