RobotLocomotion/drake #3038: Separate input port declaration from state variable allocation in System2

david-german-tri, 2016-08-03:
If we break up `CreateDefaultContext`, it becomes possible for state allocation to inspect the connected input ports, which in turn enables Systems to inherit dimensions from upstream Systems.  That might be nice.

See discussion at https://reviewable.io/reviews/RobotLocomotion/drake/2970#-KNrG4XWFHkTplOoCEJi

@sherm1 @amcastro-tri FYI

sherm1, 2016-08-03:
:+1: for flexibility in vector port dimensions -- thanks!

I'm not sure what you mean exactly by "break up CreateDefaultContext". Do you mean that there would be more calls in the API or that the internals would be different? I like the API as-is but have been thinking of the internals as phased something like:

context <- CreateDefaultContext() :
- Allocate an empty Context.
- Allocate resources and internal wiring for input/output ports.
- Recursively offer subsystems the chance to request Context resources (state, parameters, cache entries, etc.).
- Allocate any pooled resources (like if we want contiguous memory for continuous state variables).
- Return the context.

Does that match your thinking?

david-german-tri, 2016-10-12:
This has been done in LeafSystem (over several PRs).

closed 2016-10-12 by david-german-tri