RobotLocomotion/drake #3048: perching funnel spotless failure (drake configuration issue)

baxelrod, 2016-08-03:
I'm having an issue running examples/Glider/perchingFunnel. 

This is a recent source build, with spotless enabled. 
I get the following error

```Undefined function or variable 'msspoly'.

Error in perchingFunnel (line 10)
x=msspoly('x',7);```

Am I supposed to install spotless separately as well, or does the configure script take care of that?

hongkai-dai, 2016-08-03:
You could run `checkDependency('spotless')` to add `spotless` into your MATLAB path.

baxelrod, 2016-08-03:
This might be a fairly stupid issue, but I get the following error trying
that:

``` checkDependency('splotless')
Error using checkDependency (line 443)
Don't know how to add dependency: splotless```

Upon inspection I don't see any indication of spotless being installed.

Is there anything I had to do during the build other than just enable
spotless with ccmake/cmake before building?

On Wed, Aug 3, 2016 at 3:35 PM, Hongkai Dai <notifications@github.com>
wrote:

> You could run checkDependency('spotless') to add spotless into your
> MATLAB path.
>
> —
> You are receiving this because you authored the thread.
> Reply to this email directly, view it on GitHub
> <https://github.com/RobotLocomotion/drake/issues/3048#issuecomment-237396478>,
> or mute the thread
> <https://github.com/notifications/unsubscribe-auth/AA_P22CnUDqsizUOeZyvgnq-t3huRpiSks5qcRe_gaJpZM4Jb-Hh>
> .
>

-- 
Regards,
Brian Axelrod

psiorx, 2016-08-03:
> checkDependency('splotless')

Looks like you have a typo in your checkDependency call.

hongkai-dai, 2016-08-03:
When you run ccmake, can you turn `WITH_ALL_PUBLIC_EXTERNALS` on?

hongkai-dai, 2016-08-03:
@psiorx , right, it should be `checkDependency('spotless')`

baxelrod, 2016-08-04:
Yup that fixes it.

Thanks!

Any chance that change can be merged into master so my code doesn't keep
diverging?
Would you put the call before or after the megaclear call?

hongkai-dai, 2016-08-04:
You can send a pull request on the change. I would put `checkDependency('spotless')` after the megaclear call, since we use a persistent `conf` to store if spotless is enabled, `megaclear` will delete that persistent.

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri