RobotLocomotion/drake #3053: Provide a centralized Model_Instance_ID-to-Model_Name Map

liangfok, 2016-08-03:
The Model_Instance_Name-to-Model_Name map should probably be stored in the `RigidBodyTree`. Having such a map will eliminate the need to store the model name in every modeling element like `RigidBody` and `RigidBodySensor`.

RussTedrake, 2016-12-13:
Absolutely agree.  We should also modify the lcm_viewer_load_robot message to send over those model names, so that it can be displayed properly in the scene graph.  discussed quickly w/ @patmarion

jwnimmer-tri, 2018-06-09:
Moving over to @sammy-tri per recent work like #8928's second incarnation.  I suspect you can close this either now or once the second incarnation lands, perhaps?

sammy-tri, 2018-06-25:
Closing because (1) done on `MultibodyTree` and (2) WONTFIX on `RigidBodyTree`.

closed 2018-06-25 by sammy-tri