RobotLocomotion/drake #3054: Turn off msvc-64 on pre-merge jenkins builds

jwnimmer-tri, 2016-08-03:
I propose that we should turn off `msvc-64` jenkins builds in the pre-merge matrix.  I believe that every pre-merge msvc error has always shown up either in 32 only, or else both 32 and 64 (never on 64 only).  This should substantially help conserve our Jenkins resources, since pre-merge builds are much more numerous than post-merge builds.

@david-german-tri @jamiesnape @BetsyMcPhail what do you think?

jamiesnape, 2016-08-04:
I agree. We are working with AWS to finalize a yearly budget, so make a decision on this quickly would be good. Note that individual Windows instances on average cost us 5x the price of individual Linux instances, and offer 90% build time in the first and last hours due to increased startup and shutdown overhead. They are also more unreliable.

david-german-tri, 2016-08-04:
SGTM as well.  @RussTedrake, is that OK with you?

RussTedrake, 2016-08-04:
yes.  i think msvc-32 should catch what we need.

jamiesnape, 2016-08-04:
Done.

closed 2016-08-04 by jamiesnape