RobotLocomotion/drake #3057: Drake-visualiser not displaying on OSX

naveenoid, 2016-08-04:
On launching, I get this output and then nothing : 
2016-08-04 14:41:43.978 drake-visualizer[54008:37686131] *** WARNING: Method userSpaceScaleFactor in class NSView is deprecated on 10.7 and later. It should not be used in new applications. Use convertRectToBacking: instead.

soonho-tri, 2016-08-04:
I was trying to reproduce the problem on my iMac. I do get the same warning message:

```
2016-08-04 09:27:25.463 drake-visualizer[52472:470663] *** WARNING: Method userSpaceScaleFactor in class NSView is deprecated on 10.7 and later. It should not be used in new applications. Use convertRectToBacking: instead.
```

However, the Drake visualizer is launched.

[image]

naveenoid, 2016-08-04:
How do I start debugging this issue?

liangfok, 2016-08-04:
Can you post the entire console output when you try to launch Drake Visualizer on OS X?

liangfok, 2016-08-04:
I just did a fresh clone of Drake's upstream master on my OS X machine and was able to launch Drake Visualizer despite seeing the "userSpaceScaleFactor is deprecated" warning:

```
liang@Chien-Liangs-MacBook-Pro
~/dev/drake-distro-1/build( master)$ ./install/bin/drake-visualizer
2016-08-04 12:54:26.202 drake-visualizer[24321:2236144] *** WARNING: Method userSpaceScaleFactor in class NSView is deprecated on 10.7 and later. It should not be used in new applications. Use convertRectToBacking: instead.
```

I believe @soonho-tri and I are observing the same behavior. Neither of us are able to replicate this issue.

jadecastro, 2016-08-04:
FWIW, I'm not able to reproduce it either.

naveenoid, 2016-08-05:
ok I will tentatively close it..I have no clue whats wrong. Will try to replicate on a fresh clone.

closed 2016-08-05 by naveenoid

liangfok, 2016-08-05:
In addition to using a fresh clone, I also recommend re-doing _all_ of the [from source](http://drake.mit.edu/from_source.html) instructions just in case some of the platform depdendencies & environment variable settings need updating.

soonho-tri, 2016-08-05:
> I also recommend re-doing all of the from source instructions just in case some of the platform depdendencies & environment variable settings need updating.

Just want to warn you that the OS X instructions are out-dated. See https://github.com/RobotLocomotion/drake/issues/3024. I plan to get a new MBP today (or next Monday) and I can update the documentation.

naveenoid, 2016-08-05:
But I had a perfectly functional install..i just updated my master to follow upstream and something broke at some point..

liangfok, 2016-08-05:
If you can dig up the SHA of your previously working install, we can run a git diffall to identify what may have caused things to break.

soonhokong, 2016-08-05:
@naveenoid, what do you have for `otool -L ./build/install/bin/drake-visualizer`?

naveenoid, 2016-08-05:
I tried to compare that earlier..my otool output : 

./drake-visualizer:
	/Users/naveenoid/Workspace/drake-distro/build/install/lib/libddApp.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/local/opt/qt/lib/QtOpenGL.framework/Versions/4/QtOpenGL (compatibility version 4.8.0, current version 4.8.7)
	/usr/local/opt/qt/lib/QtScript.framework/Versions/4/QtScript (compatibility version 4.8.0, current version 4.8.7)
	/Users/naveenoid/Workspace/drake-distro/build/install/lib/libctkPythonConsole.1.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/Users/naveenoid/Workspace/drake-distro/build/install/lib/libPythonQt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/Users/naveenoid/Workspace/drake-distro/build/install/lib/libQtPropertyBrowser.dylib (compatibility version 0.0.0, current version 0.0.0)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/usr/local/opt/vtk5/lib/vtk-5.10/libQVTK.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/local/opt/qt/lib/QtWebKit.framework/Versions/4/QtWebKit (compatibility version 4.9.0, current version 4.9.4)
	/usr/local/opt/qt/lib/phonon.framework/Versions/4/phonon (compatibility version 4.4.0, current version 4.4.0)
	/usr/local/opt/qt/lib/QtNetwork.framework/Versions/4/QtNetwork (compatibility version 4.8.0, current version 4.8.7)
	/usr/local/opt/vtk5/lib/vtk-5.10/libvtkViews.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/local/opt/vtk5/lib/vtk-5.10/libvtkInfovis.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/local/opt/vtk5/lib/vtk-5.10/libvtkWidgets.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/local/opt/vtk5/lib/vtk-5.10/libvtkHybrid.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/local/opt/vtk5/lib/vtk-5.10/libvtkVolumeRendering.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/local/opt/vtk5/lib/vtk-5.10/libvtkRendering.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/local/opt/qt/lib/QtGui.framework/Versions/4/QtGui (compatibility version 4.8.0, current version 4.8.7)
	/usr/local/opt/qt/lib/QtSql.framework/Versions/4/QtSql (compatibility version 4.8.0, current version 4.8.7)
	/usr/local/opt/qt/lib/QtCore.framework/Versions/4/QtCore (compatibility version 4.8.0, current version 4.8.7)
	/usr/local/opt/vtk5/lib/vtk-5.10/libvtkGraphics.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/local/opt/vtk5/lib/vtk-5.10/libvtkImaging.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
	/usr/local/opt/vtk5/lib/vtk-5.10/libvtkIO.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/local/opt/vtk5/lib/vtk-5.10/libvtkFiltering.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/local/opt/vtk5/lib/vtk-5.10/libvtkPythonCore.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/local/opt/vtk5/lib/vtk-5.10/libvtkCommon.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/local/opt/vtk5/lib/vtk-5.10/libvtksys.5.10.dylib (compatibility version 5.10.0, current version 5.10.1)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
	/usr/local/opt/python/Frameworks/Python.framework/Versions/2.7/Python (compatibility version 2.7.0, current version 2.7.0)
	@rpath/liblcm.1.dylib (compatibility version 1.0.0, current version 1.3.1)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 120.1.0)

soonho-tri, 2016-08-05:
@naveenoid, I see. I couldn't see any difference between mine and yours.

naveenoid, 2016-08-05:
Im not even sure at this stage how I need to start debugging this.

liangfok, 2016-08-05:
These are long shots but:

1. Have you tried rebooting your machine?

2. Have you tried looking at your system logs (under Application → Utilities → Console) when trying to launch Drake Visualizer?

liangfok, 2016-08-05:
Another idea: Is it possible that the Drake Visualizer window is, in fact, present but just outside of the viewable screen area? http://apple.stackexchange.com/questions/34987/window-out-of-visible-workspace-range-unreachable

soonho-tri, 2016-08-05:
> Another idea: Is it possible that the Drake Visualizer window is, in fact, present but just outside of the viewable screen area?

If this is the case, you should see the following icon in your dock:

[image]

naveenoid, 2016-08-05:
I tried rebooting yesterday and no Im pretty sure the window doesnt get launched. I will try again tonight though..I will try system logs too.

naveenoid, 2016-08-06:
Ok the window was indeed hiding..<ashamed> problem is resolved.

naveenoid, 2016-08-06:
the icon changed and hence i didnt recognize the window in the dock anymore

liangfok, 2016-08-06:
Yay!!