RobotLocomotion/drake #3059: Consider turning on hidden visibility on Linux

david-german-tri, 2016-08-04:
This will more closely mimic Windows DLL behavior.  The macros are different, but are already populated in the _export files.  There's a CMake option that controls a clang/gcc flag to do this.

Also consider turning on the flag that makes failure to find a library a hard error.

liangfok, 2016-08-04:
Can you point me to some background reading that explains what "hidden visibility" means in this context?

david-german-tri, 2016-08-04:
For gcc, see https://gcc.gnu.org/wiki/Visibility.  I assume clang is similar.  

I believe the CMake flag that controls this and similar options is `CMAKE_CXX_VISIBILITY_PRESET` (though I haven't actually tested).

http://stackoverflow.com/questions/17080869/what-is-the-cmake-equivalent-to-gcc-fvisibility-hidden-when-controlling-the-e

mwoehlke-kitware, 2016-08-23:
This is... *mostly* fixed by #3065. The remaining issue is with the MATLAB code, which appears to be missing export decorations. My initial attempts to fix it were not very successful.

Eventually I think we definitely want to take another crack at this, as it will be required to move to `matlab_add_mex` (`FindMatlab.cmake`), which forces hidden visibility. (For #3065, I worked around it by having the mex stuff revert to default visibility.)

david-german-tri, 2016-10-06:
With Windows gone, we're only going to back off hidden-visibility, not expand it.  Closing.

closed 2016-10-06 by david-german-tri