RobotLocomotion/drake #3060: Should we add style guide exception for member variable names that are mathematical objects?

liangfok, 2016-08-04:
Specifically, should we allow these variables names to not have a trailing underscore?

For context, see: https://reviewable.io/reviews/RobotLocomotion/drake/2913#-KOKXMaBdJAD0EzWAk_0

sammy-tri, 2016-08-04:
One vote against.

david-german-tri, 2016-08-04:
I also vote against.

sherm1, 2016-08-04:
I also vote against an exception for the trailing `_`. 

Occasionally a member exactly represents a mathematical object that is used in referenced theory or collocated documentation. In that case the short symbol name *may* result in the clearest code, and reviewers should consider that IMO. It should still have a trailing `_` though!

rpoyner-tri, 2016-08-04:
also vote against. The trailing underscore convention solves a problem of c++ notation and clarity that overrides any misguided attempts to "match the textbook".

liangfok, 2016-08-05:
The vote is clear. 👎

closed 2016-08-05 by liangfok