RobotLocomotion/drake #3070: Drake visualizer problems on Ubuntu 16.04

naveenoid, 2016-08-05:
The drake visualizer fails to render some of the examples and throws errors seen on the terminal. 

The behaviour can be seen in this recording : 
https://youtu.be/N7LasHpjod4

david-german-tri, 2017-01-13:
Closing due to inactivity, and because lots of people are using 16.04 without complaint.  Please reopen if this is still a problem.

liangfok, 2017-02-14:
FTR, a workaround that was known to work is to open the defective `.obj` file in MeshLab and save-as another `.obj` file.

GianMassimiani, 2017-07-31:
I am having the same problem, vtkOBJReader fails at reading .obj files, raising errors such as this:

`ERROR: In /root/VTK-7.1.1/IO/Geometry/vtkOBJReader.cxx, line 320
vtkOBJReader (0x2ffbef0): Error reading 'vn' at line 14`

I am running Ubuntu 16.04. I tried with different .obj files, and also did as @liangfok suggested, opening the file in MeshLab and re-saving it. Though it didn'work. Any idea on this? @david-german-tri could you reopen the issue, please? Thanks

liangfok, 2017-07-31:
Odd. As a quick test, can you try the open/save-as technique using `view3dscene`?

GianMassimiani, 2017-07-31:
@liangfok thanks for replying. Sorry I am really a newbie on these things, I have opened the .obj file with view3dscene but there is no option to save it again as an .obj file. Only options available are .x3dv and .x3d.

liangfok, 2017-07-31:
Oops, sorry I did not realize `view3dscene` does not support save-as obj. Maybe try using `view3dscene` to convert from .obj --> .x3dv, and then using Meshlab to convert from .x3dv --> .obj? Do you know how the incompatible .obj was originally created?

jwnimmer-tri, 2017-07-31:
@liangfok Uh, if this is going to be a real bug report, then it needs a proper title and diagnosis?  The "VTK 7" in the error message makes me wonder if Gian is not using the most recent `drake_visualizer`.

@GianMassimiani Please explain exactly the steps needed to reproduce the problem.  Or at least, the steps you are trying that are not working.

GianMassimiani, 2017-07-31:
Ok, so I am using a framework called "the Director" (https://github.com/RobotLocomotion/director) which uses Drake. I get the error when using this module http://openhumanoids.github.io/director/_modules/director/ioUtils.html and in particular its `readPolyData` function. This function is calling `vtkOBJReader` with my .obj file as input, as shown in this snippet of code that I have written:

`polyData = ioUtils.readPolyData('Porsche_911_GT2.obj')`

I downloaded a few .obj files from the Internet and `vtkOBJReaded` can't read any of them, always raising an error of the type:

```
ERROR: In /root/VTK-7.1.1/IO/Geometry/vtkOBJReader.cxx, line 271
vtkOBJReader (0x2243970): Error reading 'v' at line 3
```
For example, here are the first few lines of the `Porsche_911_GT2.obj` file that is raising the above error.

```
# Wavefront OBJ exported by MilkShape 3D

v -0.830351 -0.359413 -0.750083
v -0.825420 -0.361854 -0.690273
v -0.833710 -0.363075 -0.752827
v -0.821758 -0.359107 -0.690581
v -0.832860 -0.353614 -0.068623
v -0.821758 -0.359107 -0.690581
v -0.830418 -0.350563 -0.068320
v -0.821758 -0.359107 -0.690581
v -0.832860 -0.353614 -0.068623
v -0.825420 -0.361854 -0.690273
v -0.850529 -0.360023 -0.796759
etc.
```
@liangfok As you suggested in the last reply, I tried to convert from .obj --> .x3dv with `view3dscene` and then back to .obj using `MeshLab`, but it didn't work.

GianMassimiani, 2017-08-01:
I don't know if this might be relevant,

regarding the error raised by `vtkOBJReader.cxx`:

```
ERROR: In /root/VTK-7.1.1/IO/Geometry/vtkOBJReader.cxx, line 271
vtkOBJReader (0x2243970): Error reading 'v' at line 3
```
I searched for that `vtkOBJReader.cxx` file, but I realized there is no `VTK-7.1.1` directory inside `/root`. Also, is there any good reason why VTK-7.1.1 should be installed inside /root?

liangfok, 2017-08-01:
I wonder if the problem has to do with Windows vs. Unix line endings. Can you confirm that your .obj file uses Unix line endings (\n instead of \r\n)?

Since this issue appears to be Director-related and not Drake-related, I'm going to close this issue. Feel free to open a new issue in [Director's issue tracker](https://github.com/RobotLocomotion/director/issues).

closed 2017-08-01 by liangfok

liangfok, 2017-08-01:
FTR, another source of the problem is locale settings that result in a comma being used instead of a period in floating point numbers. See https://github.com/RobotLocomotion/director/issues/541.