RobotLocomotion/drake #3075: Unable to launch Drake Visualizer Using rosrun

liangfok, 2016-08-05:
# The Error

```
$ rosrun drake drake-visualizer 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/liang/dev/drake_catkin_workspace/src/drake/externals/director/distro/pods/drake-distro/pod-build/src/director-build/lib/python2.7/dist-packages/director/drakevisualizer.py", line 3, in <module>
    from director import lcmUtils
  File "/home/liang/dev/drake_catkin_workspace/src/drake/externals/director/distro/pods/drake-distro/pod-build/src/director-build/lib/python2.7/dist-packages/director/lcmUtils.py", line 1, in <module>
    import lcm
ImportError: No module named lcm
```

# Initial Analysis

It appears that LCM's python bindings are not being correctly installed when building Drake using Catkin.

# Temporary Workaround

Start Drake Visualizer directly without using `rosrun`:

```
$ cd drake_catkin_workspace
$ ./install/bin/drake-visualizer
```

jhoare, 2016-08-09:
This sounds somewhat familiar to #1786 as there were issues requiring PYTHONPATH modifications.

liangfok, 2017-04-12:
With Drake's impending switch to being built by `Bazel`, the way in which Drake + ROS is integrated will have to change. I'm thus going to close this issue while that more drastic change occurs.

closed 2017-04-12 by liangfok