RobotLocomotion/drake #3081: Update Drake's URDF Parser to Treat World Links as part of the Model and Not Anything Special

liangfok, 2016-08-08:
Drake's URDF parser currently treats bodies called "world" as a special rigid body that represents the world within a `RigidBodyTree` (see: https://github.com/RobotLocomotion/drake/issues/2168).  This unfortunately, does not match the semantics used by RViz, which I believe is "official" since RViz is made by the same community that invented URDF. Unlike Drake, RViz treats "world" links like any other link in the model, as shown in the screenshot below:

[image]

The screenshot shows what happens when a [URDF of the Stata garage](https://github.com/RobotLocomotion/drake/blob/master/ros/drake_cars_examples/models/stata_garage_p1.urdf), which has a "world" body, is loaded into RViz. Note that the "world" body within the URDF model is treated just like any other body in the model, as indicated by the "stata_garage/world" body.

The screenshot above also shows that there is a missing transform from "stata_garage/world" to "world". This is because Drake is using an [SDF model of the Stata garage](https://github.com/RobotLocomotion/drake/blob/master/drake/examples/Cars/models/stata_garage_p1.sdf), which doesn't have a world link. We can't currently load the URDF model of the Stata Garage into Drake because Drake will eliminate the world body (since it treats it as the actual world) and abort after it detects more than one model in the URDF.

liangfok, 2016-08-08:
Note that work cannot begin on this issue until after #3064 is merged, since that PR enables multi-model support in `RigidBodySystem` and `RigidBodyTree`.

liangfok, 2016-08-12:
After exchanging a few e-mails with OSRF, the observations in this issue are correct. RViz does not maintain "world" as a keyword. Thus, a link called "world" is treated like anyother link in the URDF. Because of this, I believe this issue should remain open.

Longer term, we should do one or more of the following:

1. Modify [Transform Broadcaster](http://wiki.ros.org/tf/Tutorials/Writing%20a%20tf%20broadcaster%20(C%2B%2B)), [RViz](http://wiki.ros.org/rviz), and the rest of ROS to support SDF.

2. Extend SDF to support the inclusion of URDF models as first-class elements.

3. Extend URDF to include `<world>` as a first class tag.

liangfok, 2016-08-12:
Here are some more notes:

Gazebo uses keyword "world". Here's an example joint attaching link utorso to the world:

https://bitbucket.org/osrf/drcsim/src/194be8500fef81593f79607a21ee2badd9700a0e/atlas_description/urdf/atlas_harnessed.gazebo?at=default&fileviewer=file-view-default#atlas_harnessed.gazebo-12

Here is WIP on adding a `<world>` element to URDFs: https://github.com/ros/urdfdom_headers/tree/master/urdf_world/include/urdf_world

URDF could be generalized to allow multiple models in a single file. For example:

```
<?xml version="1.0" ?>
<robot name="robot_foo">
  <origin xyz="..." rpy="..."/>
  <link name="world" />
  ...
</robot>
<robot name="robot_bar">
  <origin xyz="..." rpy="..."/>
  <link name="world" />
  ...
</robot>

<robot name="robot_baz">
  <origin xyz="..." rpy="..."/>
  <link name="world" />
  ...
</robot>
```

SDF already assumes this format:

```
<world>
  <model name="m1">
  </model>
  <model name="m2">
  </model>
  ...
</world>
```

SDF also supports include directives:

```
<?xml version="1.0" ?>
<sdf version="1.5">
  <world name="default">
    <include>
      <uri>model://double_pendulum_with_base</uri>
    </include>
    <include>
      <uri>model://double_pendulum_with_base</uri>
      <pose>1 1 0 0 0 0</pose>
    </include>
    <model name="m1">
   ...
    </model>
  </world>
</sdf>
```

One high priority item is to modify the SDF standard to be able to include URDFs:

```
<world>
  <model name="m1">
     [SDF]
  </model>
  <model type="urdf" name="u1">
    [URDF]
  </model>
</world>
```

liangfok, 2016-08-12:
To weld a model to the world using SDF, one currently does the following:

```
<?xml version="1.0" ?>
<sdf version="1.5">
  <world name="default">
    <model name="pendulum_welded_base">
      <include>
        <uri>model://double_pendulum_with_base</uri>
      </include>
      <joint name="pendulum_world" type="fixed">
        <child>double_pendulum_with_base::base_link<child>
        <parent>world</parent>
      </joint>
    </model>
  </world>
</sdf>
```

A possible extension is to allow URDF XML to be placed inside the SDF XML:

```
<?xml version="1.0" ?>
<sdf version="1.5">
  <world name="default">
    <model name="pendulum_welded_base">
      [URDF XML for the model]
      <joint name="pendulum_world" type="fixed">
        <child>double_pendulum_with_base::base_link<child>
        <parent>world</parent>
      </joint>
    </model>
  </world>
</sdf>
```

Or, link to a separate URDF XML using `<include>` as so:

```
<?xml version="1.0" ?>
<sdf version="1.5">
  <world name="default">
    <model name="pendulum_welded_base">
      <include>
        <uri>file:///home/user/my_robot.urdf</uri>
      </include>
      <joint name="pendulum_world" type="fixed">
        <child>double_pendulum_with_base::base_link<child>
        <parent>world</parent>
      </joint>
    </model>
  </world>
</sdf>
```

tri-ltyyu, 2019-03-22:
Assigning to @jwnimmer-tri to see if still relevant

jwnimmer-tri, 2019-03-22:
From a quick scan of the code, I think our MultibodyPlant parser still treats "world" specially (it has checks against kWorldName which is `"world"`).  Whether the complaint in this issue description is a relevant defect or not, I'm not sure.  Maybe @sammy-tri can weigh in?

sammy-tri, 2019-03-25:
Yes, we still treat `"world"` in this fashion.  From reading the link to gazebo above, I think it might do the same thing? I'm OK with closing this issue.

jwnimmer-tri, 2019-03-25:
Alright.  I'll chalk this up as "When someone who cares about this and knows their business enough to (re-)articulate what the problem is speaks up, then we can re-open and fix it at that point."

closed 2019-03-25 by jwnimmer-tri