RobotLocomotion/drake #3086: Update LinearSystem API to return int instead of size_t

liangfok, 2016-08-08:
# Motivation

We are trying to avoid using unsigned integer values. See: http://drake.mit.edu/code_style_guide.html#clarifications

# Specific Methods

``
AffineSystem::getNumStates() const;
AffineSystem::getNumInputs() const;
AffineSystem::getNumOutputs() const;
```

Note that the above methods actually cast signed integer values into an unsigned values!

jwnimmer-tri, 2016-08-08:
These methods are per the System1 API that uses `size_t` for the _System Concept_.  I think our time is best spent deleting System1, rather than gold plating it.

liangfok, 2016-08-08:
Ah. I didn't realize these were specified by System 1.

closed 2016-08-08 by liangfok

jwnimmer-tri, 2016-08-08:
Yup.  That's one of the downsides of using concepts instead of interfaces.