RobotLocomotion/drake #3088: Modify RigidBodyTree to Assign World Body to have Model Instance ID Zero

liangfok, 2016-08-08:
# Motivation

The "world" is a distinct model instance. It should not be considered part of any other model in the `RigidBodyTree`.

# How?

Modify the constructor of `RigidBodyTree` to set the model instance ID of the `RigidBody` representing the world using the following code:

```
b->set_model_instance_id(add_model_instance());
```

# Why Wasn't the Above Change Merged Into Master Already?

For some reason, setting the world to have its own model instance ID [broke some unit tests](https://drake-jenkins.csail.mit.edu/job/experimental/3454/). One example is:

```
$ ctest -VV -R testIKMoreConstraints
...
153: Restoration phase is called at point that is almost feasible,
153:   with constraint violation 2.220446e-16. Abort.
153: x = -0.026      0  0.844      0      0      0      0      0      0   0.27      0  0.055  -0.57   1.13  -0.55 -0.055  -1.33  2.153    0.5 0.0985      0 0.0008  -0.27      0 -0.055  -0.57   1.13  -0.55  0.055   1.33  2.153   -0.5 0.0985      0 0.0008 0.2564
153: qsc = 0.125 0.125 0.125 0.125 0.125 0.125 0.125 0.125
153: info = 100
153: /home/liang/dev/drake_catkin_workspace/src/drake/drake/systems/plants/test/testIKMoreConstraints.cpp:189: Failure
153:       Expected: info
153:       Which is: 100
153: To be equal to: 1
153: 0.000000
153: 0.000000
153: 0.000000
153: /home/liang/dev/drake_catkin_workspace/src/drake/drake/systems/plants/test/testIKMoreConstraints.cpp:198: Failure
153: Value of: CompareMatrices(com, Vector3d(0.074890, -0.037551, 1.008913), 1e-4, MatrixCompareType::absolute)
153:   Actual: false
153: Expected: true
153: [  FAILED  ] testIKMoreConstraints.IKMoreConstraints (99 ms)
```

sherm1, 2016-08-08:
:+1: This makes good sense to me. I like the idea of dedicating model instance 0 to World. Separately, we should also make sure that *rigid body* 0 is reserved for the immobile ground body that is coincident with World's "body" 0.

amcastro-tri, 2016-08-10:
Maybe this is not the right place but @sherm1's comment made me think about it. What if the "world" actually is a non-inertial frame? for instance, a robot in a rocket with mission to Mars. In that case it would not be true then that the ground is immobile.

sherm1, 2016-08-10:
> What if the "world" actually is a non-inertial frame?

World *must* be an inertial frame (by definition) because we need an inertial frame in which to calculate accelerations. For a mission to Mars we could relocate the world frame periodically to keep location coordinates small. I think we could even arrange for the world frame to translate at a constant velocity but it can't rotate or accelerate without breaking the equations of motion; those changes would be done instead with periodic corrections.

So "World" should be defined as some inertial frame chosen to permit calculation of dynamics. If there is a more natural non-inertial frame F in which to express the system, frame F can be given prescribed motion with respect to World (including accelerations) and the rest of the system can be written with respect to F. But physically, forces would be required to make F move that way; prescribed motion would calculate those forces.

jwnimmer-tri, 2016-12-22:
@RussTedrake are you working on this?  Is it done?

SeanCurtis-TRI, 2016-12-28:
I would be sorely surprised if this is *not* the case.  [this code](https://github.com/RobotLocomotion/drake/blob/master/drake/multibody/rigid_body_tree.cc#L109) initializes the tree with the world body at zero.  Both parsers treat "world" special.  And, hardly a proof, my code relies on this being true.  :)

I strongly suspect this can be closed.

liangfok, 2016-12-28:
Actually, I believe this is still a WIP. @RussTedrake has PR #4492, which will address this issue once it's merged.

Note that this issue requires that the first model instance added to the tree not have a model instance ID of zero. Instead, it should have a model instance ID of 1.

SeanCurtis-TRI, 2016-12-28:
Ahhhh...my brain filled in words that weren't there.  I interpreted everything w.r.t. body ids and not model instance ids.  My bad.

RussTedrake, 2017-02-03:
Despite having it mostly resolved in a branch, I think this is probably best left to MBTree.  I will close this.  But fyi @amcastro-tri.

closed 2017-02-03 by RussTedrake

amcastro-tri, 2017-02-03:
Thank you Russ. Added to the feature list in #5013