RobotLocomotion/drake #3089: Add a "Planar" Floating Base Type

liangfok, 2016-08-08:
# Motivation

We want to simulate mobile robots with holonomic bases but don't want to model all of the complex contact interactions between the omnidirectional wheels and the ground. The current integrators in C++ Drake result in very high reaction forces from these contacts, resulting in the need to run simulations at excessively small time steps, which results in very low real-time factors. By switching to a dedicated floating base type that only restricts lateral movement along the Z-axis, and rotational movement around the X and Y axes, we can eliminate all contact forces with the ground and thus speed up simulations.

# Implementation

See `drake-distro/systems/plants/joints/DrakeJoint.h`.

sherm1, 2016-08-08:
:+1: A planar joint should be in every multibody toolbox -- it has many uses. For simplified ground contact it can only be used if you can be sure there will be no Z-axis liftoff.

In many cases we can recover the contact forces even with the planar joint in place -- they are represented by the joint's reaction forces which can be calculated almost for free.

jwnimmer-tri, 2017-03-11:
@liangfok Should this be reassigned to @amcastro-tri?

liangfok, 2017-03-11:
Yes, I think that's a good idea. @amcastro-tri, feel free to escalate to @sherm1 if you're the the right person for this.

sherm1, 2017-03-11:
Definitely something for the new MultibodyTree, not a retrofit for RBTree. However, this joint is easily emulated now in RBTree using a sequence of two prismatic and one revolute joint, using two dummy massless bodies to serve as intermediate frames.

jwnimmer-tri, 2019-03-26:
The same work-around applies for MultibodyPlant -- we'll close this until someone wants to advocate that the work-around is unsatisfactory.

closed 2019-03-26 by jwnimmer-tri