RobotLocomotion/drake #3096: CDash build graph axes are confusing

david-german-tri, 2016-08-09:
Example: https://drake-cdash.csail.mit.edu/buildSummary.php?buildid=74284

There are left and right y-axes, both of which are labeled in minutes.  The left axis (I think) corresponds to the curves with actual time dimensions.  The right axis (I think) corresponds to the curves that are dimensionless, and the "minutes" label just shouldn't be there.

Given that the scales are not similar even among curves that have the same units, I think it would be more useful just to show four separate graphs.

jamiesnape, 2016-08-09:
This is built into CDash, so would need fixing upstream.

david-german-tri, 2016-08-09:
Makes sense. Last time we talked about how to report bugs to Kitware products, you suggested filing them here as a starting point, but I'm happy to file upstream directly if that would be better.

jamiesnape, 2016-08-09:
I will pass it on upstream.

david-german-tri, 2017-04-11:
It looks like this fix has been rolled out on drake-cdash.  @BetsyMcPhail, please close if you agree.

closed 2017-04-11 by BetsyMcPhail