RobotLocomotion/drake #3109: The Gain system should not have to be created for a specific length

amcastro-tri, 2016-08-10:
This is in response to [this](https://reviewable.io/reviews/robotlocomotion/drake/3097#-KOkkZThFMPzpqCv9azA) comment. 

@david-german-tri's PR #3102 introduces the ability to register input/output ports without specifying their lenghts (I believe?). This issue can easily be solved after the merge of #3102.

david-german-tri, 2016-08-10:
#3102 isn't quite sufficient.  For a system like `Gain` or `Adder`, there's an implied relationship that the output size is the same as the input size(s).  Inside `Gain::AllocateOutput`, we don't have access to that information; and even if we did, it would take a handwritten implementation to make use of it.

So, I propose landing `Gain` without auto-size for the moment.  I have some ideas for how to make it work but I don't want to cram them into #3102 or to block you on them.

amcastro-tri, 2016-08-11:
I see. In R1 of #3102 I saw you had the method `System<T>::declare_continuous_input_port()` which I understood as adding a port with a length to be determined from its connections. Though now I see you don't have it anymore. I guess this is exactly because the capability to determine ports lengths at runtime (or at `Finalize()` time) is not there yet as you explain?

david-german-tri, 2016-08-11:
> I guess this is exactly because the capability to determine ports lengths at runtime (or at Finalize() time) is not there yet as you explain?

Yep, I realized it didn't make sense yet and took it out.

david-german-tri, 2016-08-11:
Thinking about this more, I think the real question is: what does it mean for an output port to be "auto-size"?  I see two possible answers offhand:

1. All the vector-valued input ports must be the same size, and the output port is that size.

2. The System author provides a function that takes the input port sizes, and returns the size of the output port.

Either of these solutions is feasible, and both require a little bit more code in `LeafSystem`.  Thoughts about which one is better?  cc @sherm1 also.

sherm1, 2016-08-11:
I think it might be helpful to think this through in more general abstract value terms. In general an OutputPort produces an AbstractValue object (these can be *very* abstract, like a point cloud). We provide convenient special-case support for when the AbstractValue is actually a numeric `Vector<T>`, and we also support (but don't require) that the Vector must have a particular size.
Non-numerical blocks like Simulink's [Multiport Switch](http://www.mathworks.com/help/simulink/slref/multiportswitch.html) can work just with the abstract value, without ever knowing that it was actually a Vector. 

One decision we need to make is at what point an OutputPort's value type must be known. Simulink's Multiport Switch appears to allow different input types, which would imply that the output type couldn't actually be known until run time when the control signal is available to choose one of the inputs. Personally I think that is too late! I would like to propose that we declare that all types must be known, and all type checking be performed, during the diagram `Finalize()` call (that is, with no Context present).

Any thoughts on the above before going on to the sizing question?

amcastro-tri, 2016-08-11:
> All the vector-valued input ports must be the same size, and the output port is that size.

That would depend on the particular system. For instance, the example @sherm1 gave before of the [Switch block](http://www.mathworks.com/help/simulink/slref/switch.html) is a good example with two input ports (required to be the same size) and a "control" port which does not necessarily need to have the same type as the two other inputs.

> The System author provides a function that takes the input port sizes, and returns the size of the output port.

That seems a plausible solution to me. That method could check if the input port sizes make sense for the particular system. For the [Switch block](http://www.mathworks.com/help/simulink/slref/switch.html) for instance you would check that the two "input signal" ports actually do have the same length.

david-german-tri, 2016-08-11:
OK, I think two different questions, both larger than the question I was originally trying to ask, have gotten mushed together here.

1. When can solvers and other external users of Systems and Diagrams assume that the input/output types and sizes are fixed?

I think @sherm1 and I agree.  Pedantically, I would say that the act of allocating a Context finalizes the input types and sizes.  The act of allocating an Output finalizes the output types and sizes.  In normal usage, both of those things happen inside `Diagram::Finalize`.

2. What degrees of freedom should the topology specification provide?

I think we all agree that "auto-size" vector ports should be possible.  An auto-size vector input port is determined by the size of the output port to which it's connected.  An auto-size vector output port is determined by a process TBD.  (The nature of that process was my original question.  Having read @amcastro-tri's response, I agree with him: the System author will have to write some code.)

It sounds like @sherm1 is also suggesting that "auto-type" ports should be possible, where the type of an input port (kVectorValued vs kAbstractValued) is determined by the output port to which it's connected.  Is that right?  If so, I am skeptical - I suspect use cases like the Switch are rare enough that the additional notation is not justified, versus just having a VectorSwitch and an AbstractValueSwitch in the primitive library.

sherm1, 2016-08-12:
> It sounds like @sherm1 is also suggesting that "auto-type" ports should be possible

Not exactly. What I'm suggesting is that all ports have AbstractValue types. Some of those have a concrete type of VectorInterface, for which we provide some extra support. For a switch, we don't need to look at the concrete type at all since we can just shuffle the appropriate AbstractValue along -- let some other block that cares complain if it isn't a Vector. But I was proposing above that we limit type flexibility to allow verification at System::Finalize(). I believe that implies for a Switch that all the inputs have to have the same AbstractValue type, so that the output can be known to have that type also. (This requires that AbstractValue have an `is_same_type_as(other_abstract_value)` method.)

sherm1, 2016-08-12:
> I think @sherm1 and I agree. Pedantically, I would say that the act of allocating a Context finalizes the input types and sizes. The act of allocating an Output finalizes the output types and sizes. In normal usage, both of those things happen inside Diagram::Finalize.

I wish I could agree completely! I do at least agree that Diagram::Finalize() is the right place for checking the validity of the wiring diagram. But I don't get how it has anything to do with allocating Context or Output objects -- I would presume those allocations would be *consumers* of type information, not the generators of it.

Meta question: is there any reason we have to distinguish "type" and "size" for thinking about diagram validity-checking? A fixed-size vector is logically a type, a kind of vector, which is a kind of abstract value. So it might be sufficient just to think in terms of type checking.

david-german-tri, 2016-08-12:
> I do at least agree that Diagram::Finalize() is the right place for checking the validity of the wiring diagram. But I don't get how it has anything to do with allocating Context or Output objects -- I would presume those allocations would be consumers of type information, not the generators of it.

I still think we're agreeing; I just confused the matter by emphasizing some implementation details.  During Diagram::Finalize, we can check the validity of the wiring diagram, which means we can compute the size (and perhaps type) of subsystem ports that were set to auto/inherited in the `SystemPortDescriptor`, which in turn means we can allocate actual hunks of memory for the port data.  Those hunks of memory are in the subsystem Context and Output.

> What I'm suggesting is that all ports have AbstractValue types. Some of those have a concrete type of VectorInterface, for which we provide some extra support.

> A fixed-size vector is logically a type, a kind of vector, which is a kind of abstract value. 

Oh, I see.  I agree this would work, and it's elegant.  However, I'm a little uneasy about making VectorInterface a second-class citizen to AbstractValue, considering that Drake has never previously had abstract-value-like entities, and using them sacrifices much of the value of Drake.

sherm1, 2016-08-12:
> I'm a little uneasy about making VectorInterface a second-class citizen to AbstractValue ...

I don't agree that being one of the possible value types makes VectorInterface a second-class citizen, any more than a Sphere is a second-class citizen of a Shape! But actually it did occur to me when I implemented abstract ports in sys3 that it might be convenient to raise the citizenship level to match the existing `Value<T>` like this:
```c++
AbstractValue
    Value<T> : AbstractValue // T is arbitrary
    VectorInterface<T> : AbstractValue // T must satisfy Eigen's scalar concept
```
The main practical issue I encountered though was that AbstractValues must be copyable and assignable but VectorInterfaces are not, so I had to create an "object" wrapper class in order to be able to instantiate `Value<VectorObject<T>>`.   

> considering that Drake has not previously had abstract-value-like entities, and using them sacrifices much of the value of Drake

That is not how I see these at all -- they *extend* the value of Drake. All the glass-box functionality is available still where the abstract values are vectors, but a great deal of useful non-numeric functionality becomes available also. And in those contexts where concrete type doesn't have to be known, being able to work uniformly with abstract types is very convenient, makes for a tidy implementation, and allows for simpler explanations.

david-german-tri, 2016-08-16:
OK, I'm convinced.  My first hack at #3164 shows that the handling abstract values and vector values separately in `InputPort`/`OutputPort` has ugly side effects that can't be concealed from user code.  Your way it is!

sherm1, 2016-08-16:
:+1: cool! Yeah, the same thing happened to me when I was hacking around in sys3.

david-german-tri, 2017-04-13:
@sherm1, I've reassigned this to you.  There has been much less demand for this feature than we anticipated last summer; if I were you I'd consider closing it as wontfix.

tri-ltyyu, 2019-03-22:
Still relevant. Keeping open

RussTedrake, 2020-08-31:
@sherm1 -- i'm not sure there is anything left to do on this one?  consider closing it?

sherm1, 2020-08-31:
Yup, I think we've lived with this well enough to say it doesn't seem to be a problem. Closing as wontfix -- someone can open a new issue if a real use case arises.

closed 2020-08-31 by sherm1

jwnimmer-tri, 2020-08-31:
What about this kind of thing still sprinkled throughout the code?

https://github.com/RobotLocomotion/drake/blob/f789501e942ea2935a0d56c2bd42576b0df8589b/systems/framework/framework_common.h#L77-L80

Seems like we should remove that false hope.  Re-open the issue to track removing it?

sherm1, 2020-08-31:
Well, I don't think it would be impossible to do this. We just don't have a use case at the moment so I don't think we need an open issue. I'm OK with leaving the thought alive in the code.

PR #13981 (2020-09-01): framework: Deprecate kAutoSize
The mention of an unimplemented feature is a distraction for users.

In pydrake, remove it immediately.  (As of Python 3.6, there is no official support for deprecating attributes; it's new as of 3.7.) This is a breaking change.

Closes #3109.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13981)