RobotLocomotion/drake #3114: Add a method to RigidBodyTree that Computes the Generalized Gravity

liangfok, 2016-08-10:
Right now we're using `RigidBodyTree::inverseDynamics()`, which is inefficient.

One method may return the gravity of all models in the RigidBodyTree (the key of returned map is the model instance ID):
```
std::map<int, VectorXd> ComputeGeneralizedGravity();
```

Another may be to take a specific model instance ID:

```
VectorXd ComputeGeneralizedGravityOfModel(int model_instance_id);
```

sherm1, 2016-08-11:
@liangfok, I think you might be mixing physical and conceptual objects here. If we have a hand model that has finger submodels, gravity affects each body independently but projecting it into joint space couples them. There is only one physical system in RigidBodyTree; sometimes you can find decoupled subsystems in it but that is not implied by the models & instances scheme we have been discussing.

I don't think we should consider model instance ids as anything other than labels indicating where certain elements came from originally. They should not be assumed to say anything about the physical structure of the resulting multibody system. Generalized gravity is not a model element so can't (in general) be unambiguously associated with any particular model instance.

liangfok, 2016-08-11:
I think it should be possible to identify just the portion of the `RigidBodyTree` that the user is interested in and to only compute the gravity compensation force / torques for the joints in that portion. Are you saying this is not worthwhile doing, or that it is not possible? What if the portion of the tree is rooted to the world?

hongkai-dai, 2016-08-11:
@sherm1 , I thought inverseDynamics function computes
```
H(q)*vdot+C(q,v)*v+g(q)
```
and when we use it to compute the generalized gravitational vector, we substitute `vdot` and `v` with 0. Thus the computation of `H(q)*vdot+C(q,v)*v` is redundant. We could create a method to compute the gravitational vector `g(q)` directly.

liangfok, 2016-08-11:
Ideally, to avoid code duplication, `RigidBodyTree::inverseDynamics()` should call `RigidBody::ComputeGeneralizedGravity()` to get `g(q)`.

sherm1, 2016-08-11:
> Thus the computation of H(q)*vdot+C(q,v)*v is redundant. 

@hongkai-dai, I agree. I was only objecting to the use of `model_instance_id` here to attempt to generate generalized gravity for a subset of the objects in the RigidBodyTree world. Using inverse dynamics with zero acceleration works but is a waste of CPU cycles.

sherm1, 2016-08-11:
> to avoid code duplication, RigidBodyTree::inverseDynamics() should call RigidBody::ComputeGeneralizedGravity() to get g(q).

That is not necessarily correct, @liangfok. That implies that there must be a calculation of generalized gravity in inverse dynamics, but that is not necessary. Instead, all the body forces (from gravity and other effects) should be collected, then transformed together into generalized forces.

sherm1, 2016-08-11:
> Are you saying this is not worthwhile doing, or that it is not possible? What if the portion of the tree is rooted to the world?

I'm saying it is not possible *in general*. If there are distinct subtrees, you can compute gravity for them separately. It *might* be the case that model instances are separate trees but we have been careful not to assign physical meaning to model instances -- in fact that is why we require all of them to be put together in a single `RigidBodyTree` object. That's because they might be coupled in arbitrary ways. We have made no statement about how model instances have to be modeled physically, and have made no limitations on how they may be coupled. You introduced that concept here in this issue, by implying that a model instance id has a separable Jacobian so that we could map its gravity forces onto its own joint dofs with no coupling to any other RigidBodyTree objects. That's what I'm objecting to.

We have agreed so far that model_instance_id can be used as *labels* for RBTree elements, so that an end user can determine what the source was for those elements. Let's just stick with that for now.

sherm1, 2016-08-11:
P.S. I'm not disagreeing with the main substance of this issue, which is that you shouldn't have to call inverse dynamics to obtain gravitational forces in joint space. BTW, you can't necessarily do inverse dynamics on individual model instances either.

liangfok, 2016-08-11:
I would like to push more "smartness" into Drake's APIs for the benefit of higher level applications. If the application wishes to know the gravity compensation torques of a 7-DOF manipulator, but the world contains hundreds of other objects, I think the method should be "smart" enough to optimize itself and only compute the minimal size Jacobian necessary to return what the application desires. Perhaps this is jumping-the-gun in terms of not solving an immediate problem. If so, lest table such a method and initially add a method that computes the generalized gravity compensation vector for everything in the `RigidBodyTree`:

```
VectorXd ComputeGeneralizedGravity();
```

I've updated the original description of this issue to prioritize a whole-tree method.

sherm1, 2016-08-11:
Good, that's better - thanks. But the API design could be different and minimally sufficient, like this:
1. First obtain spatial forces (meaning wrenches, forces and torques applied to bodies),
2. Then map spatial forces to joint forces.

Now if you are interested in gravity, you can call an API that returns gravity forces on bodies. Then map those to joint forces. If you have other external forces to deal with, you can get those, combine with gravity forces, and map once to joint space. And, that mapping to joint space is the same as multiplying by the transpose of the system Jacobian, which is a very useful thing to do in many contexts so step 2 is independently useful.

You can always determine the body forces acting on a subset of bodies (e.g., those with a particular model_instance_id). However, you cannot always form a Jacobian for an arbitrary subset of bodies, because that depends on the topology of the tree system used to represent them. But now we can reason clearly about when you might be able to get a Jacobian for a subset of bodies without having to think about gravity at the same time. That reasoning would apply to many other things you might want to do with a subset of elements, but it has nothing to do with gravity.

liangfok, 2016-08-11:
OK would the API look something like the following (I'm not sure what data structure should be used to convey spatial forces)?

```
/**
 * Returns the spatial forces applied to each rigid body in the `RigidbodyTree`. The
 * spatial forces of each body are expressed in the body's coordinate frame.
 */
std::vector<Vector6d> ComputeSpatialForces()

/**
 * Maps generalized spatial forces to generalized joint gravity forces. Returns
 * the generalized gravity forces.
 */
VectorXd ComputeJointGravityForces(std::vector<Vector6d> spatial_forces)
```

sherm1, 2016-08-11:
Something like this using a more System2-ish terminology but ignoring template argument:
```c++
/** Get spatial forces on each body due to gravity, in the configuration given
in the Context. **/
const std::vector<Vector6d>& EvalGravityForces(const Context&) const;  // #bodies x 6

/** Map a set of spatial forces on each body to the equivalent generalized
(joint space) forces. **/
VectorXd MapSpatialForcesToGeneralizedForces
    (const Context&, const std::vector<Vector6d>& spatial_forces) const;
```
With those it would be possible to write a "convenience" method like the one you originally proposed:
```c++
VectorXd CalcGeneralizedGravityForces(const Context& context) const {
  return MapSpatialForcesToGeneralizedForces(context, EvalGravityForces(context)):
}
```

tkoolen, 2016-08-22:
> Right now we're using RigidBodyTree::inverseDynamics(), which is inefficient.

It's not *that* inefficient. If you set `include_velocity_terms` to `false`, the only unnecessary computation is multiplying the motion subspace by the acceleration vector (which is known to be zero). One quick option is to add another boolean option `include_acceleration_terms` that handles this case, and then have `EvalGravityForces` delegate to `inverseDynamics` with `include_velocity_terms` and `include_acceleration_terms` set to false. You'd still have to pass in a dummy argument for `vd`, but maybe that's not terrible.

Note by the way that `inverseDynamics` is currently broken for `vd` not equal to zero (see https://github.com/RobotLocomotion/drake/issues/2201).

> Ideally, to avoid code duplication, RigidBodyTree::inverseDynamics() should call RigidBody::ComputeGeneralizedGravity() to get g(q).

That would be inefficient, because in that case you have to loop through the bodies one more time. In recursive Newton-Euler, you typically treat gravitational forces as an acceleration of the base while computing the net wrenches on the bodies, allowing you to handle the contributions of joint accelerations, joint velocities, and gravity in one pass.

liangfok, 2016-08-22:
@tkoolen: Thanks for the advice! I will do as you suggest and make `EvalGravityForces()` call `inverseDynamics()`.

EricCousineau-TRI, 2018-12-06:
Saw this in a discussion; I think this can be closed for now, since there was a workaround and we're going to deprecate RBT.

closed 2018-12-06 by EricCousineau-TRI