RobotLocomotion/drake #3119: solvers::detail::FunctionTraits does not document it's API

sammy-tri, 2016-08-11:
#3107 -- A class which conformed to the requirements of the FunctionTraits concept (which arguments to OptimizationProblem::MakeCost must conform to) was flagged in a review that it should refer to the API documentation for the concept which it implements.  The brief documentation is solvers/Function.h does not seem to meet the reviewer's requirements.

jwnimmer-tri, 2016-08-12:
This can probably be closed, given the eventual resolution within the PR.  Also I think `Function.h` is going away?

sammy-tri, 2016-08-12:
I'm not sure what would replace the code in OptimizationProblem which uses Function.h...   but if you're happy with the outcome from the original review, I'll close this.

closed 2016-08-12 by sammy-tri

jwnimmer-tri, 2016-08-12:
Oops, I think I mis-remembered the discussion about `Function.h`; the conclusion was "Yes, it's solver-specific, not common" and not "it will be replaced with FunctionalForm".

In any case, I personally don't need more documentation for it, nor in any case do I think issues are a good way to track "please document this class" :wink:.