RobotLocomotion/drake #3128: IPOPT needs tuned to pass IK tests

sammy-tri, 2016-08-11:
Currently IPOPT doesn't pass the MATLAB testIK or the C++ and MATLAB implementation of testIKtraj.  I don't think it's a fundamental limitation of the solver, but rather just some tuning parameters.  If we want an open-source solver which works for most of our IK test suite, we'll need to figure out how to tune IPOPT.

tkoolen, 2016-08-30:
testIKMoreConstraints should be added to the list after https://github.com/RobotLocomotion/drake/pull/3300 was merged.

tkoolen, 2016-10-07:
Related: https://github.com/RobotLocomotion/drake/pull/3712.

sammy-tri, 2017-06-28:
Also, if IPOPT tuning finally becomes a thing, consider fixing the trajectory optimization test which got nerfed in #6440

hongkai-dai, 2019-03-25:
Close this issue as we don't have testIK matlab code any more.

closed 2019-03-25 by hongkai-dai