RobotLocomotion/drake #3129: Add support for Bazel build system

jwnimmer-tri, 2016-08-11:
The [Bazel](http://bazel.io/) build system is a relatively new tool that provides correct, reproducible, fast builds.  It is the open-source version of google's own build tool.

I believe that Bazel would solve many of TRI's challenges in using and supporting Drake, and thus we should experiment with adding Bazel support into Drake.  For starters, it should be enough to merely add Bazel support for the newer C++ core of Drake (common, math, systems_framework).

This would start as workstation-only test, then graduate to a post-merge build only, and not in the officially supported set.  If and when we demonstrate that it is reliable and useful for developers, we can consider making Bazel support official.

It would be start on this soon, in order to help prepare and test the upcoming Bazel 0.5 official support for Windows in a few months, so that we can help drive it to meet our needs.

david-german-tri, 2016-08-11:
I'm in.

jamiesnape, 2016-08-12:
1. CI system would need to be completely refactored.
2. Multiple build systems are a bad idea.
3. Adding features to CMake that you may need in future is easy, for Bazel?

mwoehlke-kitware, 2016-08-16:
Some critical features that I'm not seeing offhand in Bazel:

- Ability to *locate* and use external libraries. ("Use" seems to be there, at least partly, but "locate" is missing.)
- Ability to install things.
- Any Windows support at all. (It looks like Bazel may assume a GCC-like compiler.)

To fully take advantage of it, we'd probably also have to port all of our dependencies.

jwnimmer-tri, 2016-08-16:
> Multiple build systems are a bad idea.

This is a compelling argument against attempting a port.  Supporting two systems in parallel during the transition will be somewhat painful.  We can perhaps localize the pain to only the bazel-porters (i.e., have a separate CI that is non-authoritative, and nobody but the porters care about).

Having a good plan of action here would be an important part of this pursuit (and I don't have one yet).

> CI system would need to be completely refactored.

I'm not too worried about this.  It will be effort and cost, but not risk.

> Adding features to CMake that you may need in future is easy, for Bazel?

As with SCons, Bazel lets you escape into (limited) Python if you really want to, and is open-source should you need to modify the actual core.

> Most of your problems are due to legacy code and the unusual PODs conventions therein. If you rewrote the build system from scratch in any language it would be an improvement.

I agree that's a big part of the problem, and that nuking the current build system code with a rewrite is a big win.  Still, if we stick with CMake, the lack of reproducibility and caching for builds and tests is a big hole CMake, that Bazel and SCons both fill.

> [Lacks the] Ability to locate and use external libraries. ("Use" seems to be there, at least partly, but "locate" is missing.)

Bazel often tends towards "build your libraries from source" (as we do with drake externals), which actually helps reproducibility, but you can also get required libraries from the system without much trouble.

And for "locate" (searching), I actually think that's a misfeature.  There should be one way to build Drake-the-entireity on a given platform, which means that the dependency will be in one well-known, hard-coded place.  Rooting around in system paths, the user's homedir, and finding something that "smells like" the right version of a library or tool only leads to confusing bug reports and wasted time.  Or if the developer _really_ wants a different path to some library, they can update their Bazel file to explicitly reference it.

> [Lacks the] Ability to install things.

Hmm?  Are you saying we'd have to roll your own "zip up these files into a release" logic, but in the CMake case that's already built-in?  That's fair, but doesn't seem like a huge difference, compared to the effort of ongoing build system upkeep and developer downtime.

> [Lacks] Any Windows support at all. (It looks like Bazel may assume a GCC-like compiler.)

This is in progress (http://bazel.io/roadmap.html), scheduled for a couple months out.  Probably slips a bit more, but still relatively near-term compared to the scope of this ticket.

> To fully take advantage of it, we'd probably also have to port all of our dependencies.

Yes, this is worth a bit of effort-assessment.  Dependencies that are just a pile of C++ code are easy to port (I've done it).  Dependencies that have bespoke `-DTHIS_AND_THAT` autoconf-like logic are harder, but there are ways to cope.

jwnimmer-tri, 2016-10-14:
FYI, active development prototype is ongoing here https://github.com/jwnimmer-tri/drake/commits/bazelspike.  It is able to build the automotive slice of the code.

The current "off the top of my head" plan is something like "allow `BUILD` files to be PR'd in-tree without CI yet" as we bring this up.  This lets other `bazel` workspaces use Drake-as-a-library, without Drake-as-a-project needing to adopt `bazel` for its demos, tests, etc.  Probably the next step after that is to get some kind of CI for Drake-as-a-library using `bazel`, but still rely on `CMake` for tests, demos, matlab, etc.  I plan to iterate with David's review on a _real_ plan offline, and then post here for comments.

jwnimmer-tri, 2016-10-20:
Next-up proposal... for C++ libraries that `bazel` knows about, teach CMake to obtain the list of sources from the `BUILD` file, instead of repeating the list in two places.

WIP at https://github.com/jwnimmer-tri/drake/tree/bazel-reuse2

jamiesnape, 2016-10-20:
Not sure that is going to work. I am trying to work out if you would get extra cmake re-configures when file lists have not changed or not enough re-configures when they do change.

This is all kind of reversed when the purpose of CMake is to generate the build files for other systems. There is no bazel support now, but there could be.

jwnimmer-tri, 2016-10-20:
When we discussed Bazel vs CMake a few months ago during the visit, the Kitware consensus in the room at that time was that Bazel is higher-level than CMake, and the one-feeds-the-other direction should go as I've done here.

In any case, what is the best solution in CMake for a list of library sources to be dynamically computed?  I had done https://github.com/RobotLocomotion/drake/commit/6a1a04211cdb5b5c95dc011683efacacf079a43c originally -- which asks a python program to emit a list of sources -- but the approach in the `reuse2` branch seemed like it would scale to generating even more of the listfile content directly from the BUILD.

jwnimmer-tri, 2016-10-20:
The other option I've considered is to _fully omit_ the listfiles for core Drake (common, math, systems, solvers, etc.) and just have CMake delegate to bazel to build the core.  That is probably better long term, but I was hoping to get there incrementally.

jamiesnape, 2016-10-20:
That would certainly be cleanest.

jamiesnape, 2016-10-20:
Then CMake would provide an external interface for CMake consumers, build CMake externals, and other legacy code. We have integrated with external tools like Ant and Gradle for Java before, so this may not be that different.

jamiesnape, 2016-10-21:
(Might be an idea to create a ticket for that and assign it to Kitware since we have done similar many times before.)

jwnimmer-tri, 2016-10-21:
Yeah.  I have to give more thought and discussion to the optimal path forward.  For now perhaps I'll just keep the two spellings of the list-of-sources duplicated.

jamiesnape, 2016-10-21:
Sure. My general opinion is that whatever the merits of Bazel or CMake, having two build systems duplicating each other or working in unusual ways to accommodate each other is going to cause a lot of extra maintenance at best, confusion or bugs at the worst. I also think there is some scope for upstreaming some useful Bazel support to CMake which might make things easier.

jwnimmer-tri, 2016-12-27:
Starting from the wonderful list in #4045, here's a checklist of our externals, to be ticked off once Bazel support for them is complete for Drake's use.  I will keep this updated as we make progress.

- [ ] `bot_core_lcmtypes` #4608
- [x] `bullet` #4394
- [ ] `ccd` #4600
- [ ] `director`
- [ ] `dreal`
- [x] `eigen` #3819
- [ ] `fcl` #4600
- [x] `gflags`
- [ ] `google_styleguide`
- [x] `googletest`
- [x] `gurobi` #4518
- [ ] `ipopt`
- [x] `lcm`
- [ ] `libbot`
- [ ] `mosek`
- [ ] `nlopt`
- [ ] `octomap` #4600
- [ ] `robotlocomotion_lcmtypes` #4608
- [ ] `signalscope`
- [ ] `snopt`
- [x] `spdlog` #4039
- [x] `yaml_cpp` #4494

jwnimmer-tri, 2017-01-06:
Relatedly, here's a checklist of tooling needs, to be ticked off for Bazel support.  I will keep this updated as we make progress:
 - [ ] cpplint
 - [ ] doxygen
 - [ ] sphinx
 - [ ] valgrind
 - [ ] kcov

The goal is only that they can be used without having to run CMake first.  Things like, e.g. doxygen don't necessarily have to be a Bazel target -- a shell script called by CMake than can also be called manually would be fine.

david-german-tri, 2017-01-13:
Here's a checklist of things that the Bazel build accesses non-hermetically, which should be made hermetic.

Libraries

- [ ] glib
- [ ] gurobi

Binaries
- [ ] gcc-4.9 (Linux)
- [ ] gcc-5.x (Linux)
- [ ] clang-3.9 (Linux)
- [ ] clang (OS X)
- [ ] other toolchain components

david-german-tri, 2017-01-13:
Here is a migration plan from CMake to Bazel, extracted from f2f conversation with myself, @jwnimmer-tri, and @jamiesnape.  It preserves the current CMake interface downstream.

1. 100% of C++ Drake is supported in the Bazel build, parallel to the CMake build.
2. We provide sufficient tools that all Drake developers can function without CMake.
3. We define a `drake.so` Bazel target that includes all of C++ Drake.
4. We extend `drake.so` to emit a CMake config file, so that `find_package(drake)` is possible.
5. In one PR, we switch from CMake to Bazel in Drake proper:
  1. The Drake CMake superbuild calls the Drake Bazel build as an external project, setting debug/release/san options on the `bazel` command line as appropriate.
  2. The Drake CMakeLists.txt are gutted of all references to library source code.  Instead, the existing libraries, executables, and tests will link against the results of `find_package(drake)`.
6. We drop the unit test source code from the `CMakeLists`, and instead just have ctest run tests out of `bazel-bin`.  We delete the `CMakeLists` rules for executables.
7. We remove all the externals from the Drake CMake superbuild, except for Director.
8. Once CDash or another dashboard tool is able to handle Bazel outputs, we stop running unit tests in ctest at all.  (speculative)

liangfok, 2017-01-14:
I'd like to add some notes to the above migration plan on how Drake+ROS will evolve as Drake transitions to being built by Bazel. For example, will we have a Catkin package whose CmakeLists.txt  use [execute_process()](https://cmake.org/cmake/help/v3.5/command/execute_process.html) to run Bazel to build Drake and produce a `lib_drake` and a CMake config file? Wil `lib_drake` and it's CMake config file then need to be moved to a known location so other ROS packages can use it?

jamiesnape, 2017-01-16:
Yes, use by Catkin is going to be tricky. Bazel and Catkin are polar opposites with how they use the environment of the system on which they are being built.

jamiesnape, 2017-01-16:
We still also need to reconfigure CI to checkout to somewhere other than the root of the Jenkins workspace.

david-german-tri, 2017-01-17:
> It preserves the current CMake interface downstream.

@liangfok The exact same Drake CMake library targets that have always existed, will continue to exist.  Does this suffice for the ROS/Catkin build?  At a glance, it seems like it should.

liangfok, 2017-01-17:
Oh, maybe I misunderstand the migration plan, specifically step 6b:

> The Drake CMakeLists.txt are gutted of all references to library source code. Instead, the existing libraries, executables, and tests will link against the results of find_package(drake).

Does this mean that ROS / Cakin no longer build's Drake, but rather treats Drake as a preexisting library that's already installed in the system?

jamiesnape, 2017-01-17:
Seems so, unfortunately. Not sure Catkin will be happy.

wjwwood, 2017-02-14:
Catkin is basically just CMake, so if a simple, pure-CMake example can be provided that "utilizes" drake (regardless of how drake was built, with either cmake or bazel) then you can make it work with other catkin packages. If you're talking about building drake along with many other things at the same time with something like `catkin_tools`, then we'd have to look at how bazel is exposed to other non-bazel libraries, i.e. how is it distributed. I can help with this as needed (I originally wrote `catkin_tools`) because I'd like to understand better how we could integrate with bazel projects.

jamiesnape, 2017-05-12:
**Jenkins TODO**

Mac:
- [ ] coverage
- [ ] debug
- [ ] release

Linux Trusty:
- [ ] coverage
- [ ] debug
- [ ] release

Linux Xenial:
- [ ] asan
- [ ] coverage
- [ ] debug
- [ ] lsan
- [ ] msan
- [ ] release
- [ ] tsan
- [ ] ubsan
- [ ] valgrind

jamiesnape, 2017-05-30:
** Installation TODO **

- [ ] bot_core_lcmtypes
- [ ] bullet
- [ ] ccd
- [ ] com_github_bazelbuild_buildtools *(skip)*
- [ ] com_github_gflags_gflags
- [ ] drake_visualizer
- [ ] eigen
- [ ] fcl
- [ ] fmt
- [ ] gfortran *(system)*
- [ ] glib *(system)*
- [ ] google_styleguide *(skip)*
- [ ] gtest
- [ ] gurobi *(commercial)*
- [ ] ignition_math
- [ ] ignition_rndf
- [ ] io_bazel_rules_go *(skip)*
- [ ] ipopt
- [ ] lcm
- [ ] libbot
- [ ] mosek *(commercial)*
- [ ] net_sf_jchart2d_jchart2d
- [ ] nlopt
- [ ] numpy
- [ ] octomap
- [ ] optitrack_driver
- [ ] org_golang_x_tools *(skip)*
- [ ] protobuf
- [ ] pybind11
- [ ] pycodestyle *(skip)*
- [ ] pycps *(skip)*
- [ ] python *(system)*
- [ ] robotlocomotion_lcmtypes
- [ ] sdformat
- [ ] semantic_version *(skip)*
- [ ] six
- [ ] spdlog
- [ ] tinyobjloader
- [ ] yaml_cpp

jamiesnape, 2017-06-27:
**CMake Removal TODO**

drake-superbuild:

- [ ] cmake
- [ ] externals

drake:

- [ ] automotive
- [ ] bindings
- [ ] common
- [ ] doc *(refactor)*
- [ ] examples
- [ ] geometry
- [ ] lcm
- [ ] lcmtypes
- [ ] manipulation
- [ ] math
- [ ] matlab *(refactor)*
- [ ] multibody
- [ ] systems
- [ ] thirdParty
- [ ] util

jamiesnape, 2017-06-28:
Three proposals regarding CI for PRs:
1. Switch LSan builds from CMake to Bazel.
2. Switch off standalone cpplint builds (or replace with a Bazel one, but lint runs on all Bazel builds currently, anyway)
3. Switch on at least one "everything" Bazel build.

mwoehlke-kitware, 2017-06-28:
I was thinking that a Bazel lint-only build might be useful if we could also *switch off* the lint tests on the other Bazel CI builds. This would make it easier to tell lint failures from "real" failures.

...Just a thought; feel free to hate it :smile:.

jamiesnape, 2017-06-28:
At the moment switching off lint gets ugly because `--test_tag_filters` do not accumulate intelligently.

jwnimmer-tri, 2017-06-28:
To me, its important that CI builds match user builds, so we shouldn't add or remove tests in CI.  The build variants should merely cover the matrix of supported platforms and documented command-line flags (such as `--compiler=`.)

jamiesnape, 2017-06-28:
Any opinion on the three proposals, and do you have a preference for the particular "everything" build? We can add others once CMake is turned down.

jwnimmer-tri, 2017-06-28:
> Switch LSan builds from CMake to Bazel.

Sure.

> Switch off standalone cpplint builds

Sure.

>    Switch on at least one "everything" Bazel build.

Anything Xenial & Everything would be fine by me.

jwnimmer-tri, 2018-01-31:
The checklists upthread probably need a refresh at some point.

jamiesnape, 2018-01-31:
We could probably split off the remaining Jenkins TODOs into separate issues and close this, I think. Other than those, I just see hermetic Gurobi remaining, and that does not see worth keeping the issue open for now (or necessarily even bothering about).

jwnimmer-tri, 2018-01-31:
I agree we don't need an issue tracking Gurobi stuff.  If its a problem in some context, we can fix it if /  when needed.

RussTedrake, 2018-09-23:
@jamiesnape, @jwnimmer-tri -- i think it's safe to close this?  ;-)

closed 2018-09-23 by RussTedrake