RobotLocomotion/drake #3130: OptimizationProblem does not provide infeasible_constraints

sammy-tri, 2016-08-11:
(referenced in #1667 as a SNOPT specific issue as well)

In the MATLAB implementation of NonlinearProgram and in the (now removed by #2951) direct-to-SNOPT inverse kinematics code, there was an ability to return the names of any constraints which were outside of their bounds if/when the solver failed to solve the problem.

There are some questions about restoring this functionality.

- The IK SNOPT solver only used this when the debug_mode flag was set, and it's not clear how often that's true.
- In the OptimizationProblem world, Constraints don't have names anymore (this could be added, but the defaults might be unhelpful)
- It's not completely clear how to make this work for some non-SNOPT solvers (though I think we could get most of the way there)

Should we resolve this issue?  Or remove the references to infeasible constraints from the IK API and call it closed?

wxmerkt, 2016-08-11:
For humanoid motion planning development as in openhumanoids we really liked knowing which constraints are unfeasible and use it frequently (e.g. to relax assumptions/change constraint sets/have the user decide how to change the IK request/human override = deciding that violation of that particular constraint is OK)

hongkai-dai, 2016-09-19:
I am going to add a field to `Constraint`
```
std::vector<std::string> name_
```
together with its getter and setter. The size of `name_` is the same as `Constraint::num_constraints`. The default value of constraint name is `generic_constraint`, the default name for linear constraint is `linear constraint`, etc. Then the `RigidBodyConstraint` will set the constraint name, and IK will return the names of the infeasible constraint.

RussTedrake, 2018-03-13:
I'm thinking of implementing this now.  It would be very useful.  
I'm trying to decide where to put it.  It's tempting to put it at the MathematicalProgram level, but only a subset of the solvers (e.g. snopt) actually returns a solution that is infeasible only in the problematic constraints.  So the approach we've taken to implementing it before is a bit solver specific?

EricCousineau-TRI, 2018-03-13:
My thoughts were to try and accommodate both: if the solver supports it, it could update some internal state to mention which `Constraint` objects are infeasible (mapping from rows to the constraint name, or something like that to make it user-friendly and semi-traceable to the code); if the solver does not support retrieving infeasible constraints, but does return an intermediate solution where it stopped, then go through each constraint, call `CheckConstraint`, and use this to loosely determine which constraints failed (not sure how succinct that is, but just a thought), but possibly with a loss of granularity.

RussTedrake, 2018-03-13:
and what should the result be when the solver does not return an intermediate result on failure (most of ours actually do not?).  Do we throw?

EricCousineau-TRI, 2018-03-13:
Yup, that would make sense if the user tries to request the infeasible constraint set.

That, or perhaps wrap it in `optional<>` so that they're not forced to deal with / explicitly catch exceptions, especially given that the solver will can switch based on the problem / environment.

hongkai-dai, 2018-03-13:
BTW, we have `EvalBindingAtSolution(const Binding<C>& binding)` in `MathematicalProgram` that can be useful, to evaluate the binding solution. That's actually how I manually detect the infeasible constraints now.

RussTedrake, 2018-03-13:
Right.  We might want to automatically call infeasible constraints if the solver returns infeasible constraints as a SolutionResult.  but should not throw in that case...

RussTedrake, 2019-04-01:
fyi -- i've got this mostly implemented.  will PR it shortly.

RussTedrake, 2019-04-01:
I initially posited that #11103 would close this, but there might be a follow-on activity to have it called automatically from `Solve` for compatible solvers, and printing (via sdplog)?

PR #11103 (2019-04-02): Provide GetInfeasibleConstraints
This is a long-missed feature by former developers of mathematical programs in the Matlab versions of drake.  
Resolves #3130

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11103)
commit 35136504 (2019-04-02): Provide GetInfeasibleConstraints (#11103)

This is a long-missed feature by former developers of mathematical programs in the Matlab versions of drake.
Resolves #3130
closed 2019-04-02 by RussTedrake