RobotLocomotion/drake #3133: Add GMock to the build

david-german-tri, 2016-08-12:
CMake provides `find_package(GTest)` but not, apparently, `find_package(GMock)`.  Not sure what the best approach is.

https://github.com/RobotLocomotion/drake/blob/master/drake/CMakeLists.txt#L191

jwnimmer-tri, 2017-03-05:
This should probably happen in the Bazel workspace, not the CMake superbuild.  Thus, we can do it ourselves anytime we want.

david-german-tri, 2017-04-11:
> This should probably happen in the Bazel workspace, not the CMake superbuild. Thus, we can do it ourselves anytime we want.

I agree, but we can't actually start using it until the CMake build is turned down, so I'm going to leave this issue open to track it.

david-german-tri, 2017-04-14:
Assigning to @jamiesnape for delegation as needed.  Not urgent.

jamiesnape, 2017-04-14:
Separate library to gtest, or like this: https://github.com/google/protobuf/blob/master/gmock.BUILD.

jamiesnape, 2017-04-14:
Ignore that. I guess they basically need to be together.

PR #5835 (2017-04-17): Add GMock to the Bazel build and add basic FindGMock module for CMake
Closes #3133.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/5835)
commit 26df7940 (2017-04-17): Add GMock to the Bazel build

Closes #3133.
closed 2017-04-17 by david-german-tri