RobotLocomotion/drake #3138: Mysterious CI Error: Undefined function 'hold' for input arguments of type 'char'.

liangfok, 2016-08-13:
# Example Errors

```
Undefined function 'hold' for input arguments of type 'char'.

Error in testMixedIntegerConvexDemo (line 43)
hold on
```

```
Undefined function 'hold' for input arguments of type 'char'.

Error in rrt_bugtrap (line 8)
figure(1); clf; hold on;
```

# Example Logs

* https://drake-jenkins.csail.mit.edu/job/experimental/3630/compiler=gcc,label=linux_matlab,type=release/

jwnimmer-tri, 2016-08-13:
Recall that every new matlab toolkit use hits the license server again, so a license flake can randomly show undefined functions.  Retry and move along.

closed 2016-08-13 by jwnimmer-tri

liangfok, 2016-08-13:
I wonder whether each CI server could have its own local MATLAB license to avoid these types of faults.

liangfok, 2016-08-13:
I searched through the [full console log](https://drake-jenkins.csail.mit.edu/job/experimental/3630/compiler=gcc,label=linux_matlab,type=release/consoleFull) trying to find the syntax of the MATLAB license checkout error but failed to find it. Here's what I see:

```
01:02:59 88: Matlab test  output:
01:02:59 88: 
01:02:59 88:                             < M A T L A B (R) >
01:02:59 88:                   Copyright 1984-2015 The MathWorks, Inc.
01:02:59 88:                    R2015b (8.6.0.267246) 64-bit (glnxa64)
01:02:59 88:                               August 20, 2015
01:02:59 88: 
01:02:59 88:  
01:02:59 88: To get started, type one of these: helpwin, helpdesk, or demo.
01:02:59 88: For product information, visit www.mathworks.com.
01:02:59 88:  
01:02:59 88: 
01:02:59 88: 	Academic License
01:02:59 88: 
01:02:59 88: Undefined function 'hold' for input arguments of type 'char'.
01:02:59 88: 
01:02:59 88: Error in rrt_bugtrap (line 8)
01:02:59 88: figure(1); clf; hold on;
01:02:59 88: 
01:02:59 88:  
01:02:59 88: 
01:02:59 88: CMake Error at /home/ubuntu/workspace/experimental/c5b85b52/drake/cmake/MatlabTestsRedirect.cmake:108 (message):
01:02:59 88:   [MATLAB] TEST FAILED Matlab returned 1
```

Notice that the output above includes a line stating "Academic License". I wonder what is the "magic keywords" to search for to positively identify this as a license checkout error.

jwnimmer-tri, 2016-08-13:
My understanding is that an "undefined function" for matlab code that isn't changed by the PR is likely to be a CI flake.  A "retest this please" will usually have a quick turn-around, so I'll usually do that if I'm pretty sure that I couldn't have broken the test in question -- without even reading the log in any detail.

Jerryxiaoyu, 2018-12-06:
Unfortunately, I encountered the same problems. I wonder whether I should reinstall MATLAB with a stand-alone license. 
Could you give me some suggestions? @liangfok @jwnimmer-tri