RobotLocomotion/drake #3139: Manual Triggering of ROS CI Builds Did Not Work

liangfok, 2016-08-13:
I attempted to trigger it [here](https://github.com/RobotLocomotion/drake/pull/3136#issuecomment-239617005) but the builds did not occur (or, they at least didn't show up in the PR's webpage).

https://github.com/RobotLocomotion/drake/pull/3136

liangfok, 2016-08-13:
FYI @jhoare.

jhoare, 2016-08-15:
I know we have been having some issues getting rate limited by github since we have been hitting their API too quickly. Could that be the cause of this @jamiesnape ? 

I see that it has been summoned elsewhere, such as in #3137 so it looks more like a service lapse rather than a permanent outage.

jamiesnape, 2016-08-15:
You triggered it from within Reviewable?

jamiesnape, 2016-08-15:
@jhoare Nothing to do with rate limiting. Different user and different IP address.

jamiesnape, 2016-08-15:
Either something to do with issuing a retest this please in the same request. Also, it is best to comment directly in GitHub so Jenkins does not get confused by the extra metadata inserted by Reviewable.

liangfok, 2016-08-15:
Aha! Yes, I issued the request through Reviewable. From now on, I will (1) not combine these types of requests with "retest this please", and (2) issue them directly via Github's website. I will now close this issue unless I encounter this problem again. Thanks!

closed 2016-08-15 by liangfok