RobotLocomotion/drake #3140: Implement a convenience method to access output port vectors

amcastro-tri, 2016-08-14:
In System 2.0 we'd like users to be able to say:
```c++
auto& source_->get_output_vector(context, 0);
```

where `auto` resolves to an `Eigen` type that the user can readily use in a mathematical expression.

The implementation will look something like:

```c++
auto& get_output_vector(const ContextBase<T>& context, int port_index) {
const BasicVector<T>* output_vector =
      dynamic_cast<const BasicVector<T>*>(
          output_->get_port(port_index).get_vector_data());
  ASSERT_NE(nullptr, output_vector);
return output_vector->get_value();
}
```

liangfok, 2016-08-14:
I believe:

```c++
  ASSERT_NE(nullptr, output_vector);
```

should instead be:

```c++
  DRAKE_ABORT_UNLESS(nullptr, output_vector);
```

Since that will continue to be enforced even in release builds.

amcastro-tri, 2018-08-14:
This clearly went out of date and is resolved now.

PR #3175 (2016-08-27): System accessors returning Eigen expressions
Closes #3140.

The new methods introduced in `System<T>` are:
- get_input_vector
- get_mutable_output_vector
- get_continuous_state_vector

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3175)
closed 2018-08-14 by amcastro-tri