RobotLocomotion/drake #3141: Consider using `-impl.h` rather than `-inl.h` for template implementations

sherm1, 2016-08-15:
We have been considering building up FCL to be the main collision library for Drake. I've been working with the community there to modify FCL to make that smoother, for example by removing boost and using gtest. @jslee02, the most active developer on the FCL collision library project, recently overhauled FCL to (a) convert it to use Eigen, and (b) templatize it uniformly by scalar type for "glass box" purposes (he even included an AutoDiff test). Needless to say, it now takes much longer to compile! So I pointed JS to @jwnimmer-tri's `-inl.h` design pattern in flexible-collision-library/fcl#160. He said:
> I like this idea! ... For the suffix, I would prefer `-impl.h` because `-inl.h` sounds like for inline functions.

I think he is right that "inline" is not the right term for these implementation headers. They are not necessarily inline in the C++ meaning of the word. I propose we switch to the `-impl.h` suffix also before we have too many of the other ones. It would also be nice to match FCL on this since it is adopting our design pattern!

I think this is up to you to decide, @jwnimmer-tri, so I have assigned this to you for a first look.

jwnimmer-tri, 2016-08-15:
For reference, Drake #2447 was the introduction of this concept.

I assert that `-inl.h` is the industry-standard name for such things, which is why I used it.  (Actually `.inl` is another common name, but even worse.)  I agree it's a bit odd, but on the other hand if you are including the `-inl.h`, very likely some inlining is going on!

sherm1, 2016-08-19:
`-inl.h` is the winner and FCL adopted it too.

closed 2016-08-19 by sherm1