RobotLocomotion/drake #3142: Need Utility Method That Returns Pose of Model Instance In World

liangfok, 2016-08-15:
Currently, determining a model instance's pose is difficult because it requires checking joint types and DOFs (within `DrakeJoint`), and system state vector indices (within `RigidBodyTree`). It results in writing a huge amount of code. For example, see [this](https://github.com/liangfok/drake/blob/d47e08eed79bf79824ed507b7603a7c2c5c24959/ros/drake_ros_systems/include/drake/ros/systems/ros_sensor_publisher_joint_state.h#L353). We should provide some utility methods in `RigidBodyTree` that does this work once and for all.

Here's a suggestion. 

```
/**
 * Returns true if the specified model instance is connected to the rest of the
 * `RigidBodyTree` by a floating joint.
 */
bool RigidBodyTree::HasFloatingRoot(int model_instance_id);

/**
 * Returns the pose of a model instance in Euler angles (i.e., x, y, z, roll, pitch, yaw).
 */
Vector6d RigidBodyTree::GetPoseRPY(int model_instance_id);

/**
 * Returns the pose of a model instance in Euler angles (i.e., x, y, z, qx, qy, qz, qw).
 */
Vector7d RigidBodyTree::GetPoseQuaternion(int model_instance_id);
```

liangfok, 2016-08-15:
@sherm1, previously (i.e., https://github.com/RobotLocomotion/drake/issues/3114), you objected to broadening the use of `model_instance_id` within `RigidBodyTree`. Do you object to its use as described above?

sherm1, 2016-08-16:
I'm not opposed to utility methods if they solve real problems and don't create future maintenance agony. One advantage to a method like `HasFloatingRoot()` is that we can implement it expeditiously now but could convert it to an efficient implementation later if it proves necessary, without changes to user code. If we force user code to perform that operation we can't speed it up later.

A few comments:
- What does it mean for a model to have a pose? Is that something fixed that doesn't change or something state dependent?
- For a minimally-sufficient API, there is no need to add a method for each orientation representation. Much better is to return a single standardized representation (rotation matrix or quaternion, say), with the orientation classes providing a minimally sufficient API for interconversion (e.g., rotation matrix to Euler angle sequence).

liangfok, 2016-08-16:
> What does it mean for a model to have a pose? Is that something fixed that doesn't change or something state dependent?

By this I mean the state of a (actual or effective) floating joint that connects to the model instance to the world. The _actual_ state is returned when the floating joint connects the model instance to the world. The _effective_ state is returned when the floating joint connects the model instance to some other model instance in the rigid body tree. Obviously, based on the above definition, the return value depends on the current joint states.

> For a minimally-sufficient API, there is no need to add a method for each orientation representation. Much better is to return a single standardized representation (rotation matrix or quaternion, [...]

OK good point. I will update the original proposal to use `Isometry3d` as the return value.

sherm1, 2016-08-16:
@liangfok: A model instance isn't a physical thing so doesn't connect to the world. There may be some bodies included in the model instance that are connected directly to the world. Your proposed interface introduces a requirement on model instances that they have a single "root" body -- to my knowledge there is no such requirement (certainly if the instance came from an `.sdf` file it could have many attachments to the world). Also, from the comment you now propose for `GetPoseOfModelInstance()` it is clear that you simply want to return the pose of a particular body; that capability already exists.

It sounds like what you need is a way to find out which bodies of a model instance have "world" as their parent body. Is that correct? After you find those bodies you can certainly obtain their poses using existing code.

liangfok, 2016-08-16:
@sherm1: I see, you're saying that the concept of a "root" of a "model instance" is not known within the `RigidBodyTree` and thus any bookkeeping necessary to define the "root" should remain external to the `RigidBodyTree`. This is OK with me. If the application has a pointer to a rigid body that it considers to be a root of a model instance, it can get the "pose" of the model instance by calling: `RigidBodyTree::relativeTransform()`, `RigidBodyTree::relativeQuaternion()`, or `RigidBodyTree::relativeRollPitchYaw()`. This SGTM. I shall thus close this issue.

sherm1, 2016-08-17:
> I see, you're saying that the concept of a "root" of a "model instance" is not known within the RigidBodyTree and thus any bookkeeping necessary to define the "root" should remain external to the RigidBodyTree. 

Yes, but the `RigidBodyTree` does understand the topology (tree structure) of the system so could tell you which bodies have `world` as their parent. Those are sometimes called "base bodies" by multibody dynamicists. So it would be reasonable to have that operation done in RBTree (might already be there). In general you could ask for a list of the children of a particular body. In this case you want to know who are the children of "world". That could be a general API, with an optional model_instance_id for filtering, something like:
```c++
std::vector<const RigidBody*> FindChildrenOfBody(const RigidBody&, int model_instance_id=-1) const;
```
or if base bodies are particularly interesting you could have something like:
```c++
std::vector<const RigidBody*> FindBaseBodies(int model_instance_id=-1) const;
```
Neither of those make any assumptions about the physical properties of model instances -- they just query the RBTree for things it knows about, filtering the list for convenience (and to avoid heap-allocating unwanted stuff in the return value).

liangfok, 2016-08-17:
Nice! I think both of those methods will come in handy and will thus re-open this issue with the objective of implementing them. I looked through `RigidbodyTree`'s API and neither method exists. Also, none of the existing "find" methods appear already provide what these methods will provide.

hongkai-dai, 2016-08-17:
@liangfok , I can work on this. @sherm1 , should the return argument of these two methods be `std::vector<const shared_ptr<RigidBody>>`, instead of using raw pointer?

sherm1, 2016-08-17:
> should the return argument of these two methods be std::vector<const shared_ptr<RigidBody>>, instead of using raw pointer?

@hongkai-dai: no, these should be raw pointers because they are simply references to existing objects within the RigidBodyTree -- we don't want to transfer any ownership along with them. The comments for the methods should explain that the pointers are only valid while the RBTree is still alive.

sherm1, 2016-08-17:
P.S. They should be const pointers unless @liangfok has a compelling argument for why they should be mutable.

sherm1, 2016-08-17:
Oh, I just saw in #3168 that the pre-existing `findAncestorBodies()` returns body indexes (ints) rather than pointers -- ints are better than pointers! The proposed routines are very similar so I think if you write them they should look like a matched set with the updated signature for `FindAncestorBodies()`.

@liangfok, is there any reason these methods shouldn't return body indexes also?

liangfok, 2016-08-17:
> @liangfok, is there any reason these methods shouldn't return body indexes also?

I can think of only two arguments _against_ exposing the indices and don't know how valid they are:

1. Exposing the indices also exposes the internal data structure of the `RigidBodyTree`. Clearly, for data protection reasons, `RigidBodyTree::bodies` should be a private member variable.

2. Exposing the indices may result in application errors once we support the ability to dynamically add and remove bodies from the `RigidBodyTree` at run time.

Again, I'm not sure whether the above two concerns are of any consequence or whether there are any other concerns I haven't thought of yet. Please let me know what you think!

liangfok, 2016-08-17:
I just discovered that I need this for #3103 as I convert `SensorPublisherOdometry` to support multiple model instances. I shall thus take over this issue.

liangfok, 2016-08-17:
I updated the original description of this issue with versions of the methods that work with body index values rather than pointers / references to `RigidBody` objects. This is despite the two drawbacks I mentioned above, which I think are surmountable.

sherm1, 2016-08-17:
I agree that the issues with int indexes are there but surmountable. RBTree should have a method that maps body index to `RigidBody*` -- I totally agree that the `bodies` data member should be private!

The issue of indexes being invalid after add/remove bodies dynamically can be dealt with in documentation, for example as the the STL containers do for iterators. The same problem exists for pointers anyway if we allow bodies to be deleted.

:+1: for the new signature proposals. I especially like that they match `FindAncestorBodies()`!

closed 2016-08-19 by liangfok