RobotLocomotion/drake #3143: Implement Context<T>::ReserveContinuousState

amcastro-tri, 2016-08-15:
As discussed in [this reviewable comment](https://reviewable.io/reviews/robotlocomotion/drake/3102#-KOvJLeGJGUuRJANReRx) this method will help users when overriding `LeafSystem<T>::ReserveState`

sherm1, 2016-08-22:
Sorry to beat a dead horse but I would really like to see this done top down to make sure it is actually doing what we need. In this case we need a way for each subsystem to reserve context resources for its own use -- continuous & discrete state variables, mode variables, cache entries at least. Then we need to be able to express dependencies on those quantities (as well as dependencies on time, inputs, and parameters).

Given that picture, it would be very useful to have a concept of "cache resource", and a way to keep a handle so that (for example) if I ask for a group of discrete state variables and later a cache entry, I have a way of saying "mark this cache entry as dependent on that group I reserved earlier". Currently I don't see any provision for that, so I think we're missing some important abstractions.

Without taking the full system into account, I'm afraid we will waste a lot of time implementing bits and pieces that don't actually fit into the final picture.

amcastro-tri, 2017-02-01:
@sherm1, are you happy with the current solution? can we close this? or open/reference a new one that mentions caching and other details?

sherm1, 2017-02-01:
This is already accounted for in #4364.

closed 2017-02-01 by sherm1