RobotLocomotion/drake #3147: mex segfaults in InstantaneousQPController

david-german-tri, 2016-08-15:
On both [Linux/Debug](https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=79072) and [Mac/Release](https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=79147), the following tests are failing inside constructQPDataPointerMex:

* runAtlasBalancingPerturb
* runAtlasManip
* runValkyrieBalancingPerturb

As you can see from the stack traces, the segfault happens early in `InstantaneousQPController::initialize()`, which is called from the constructor, which is called [here](https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=79147).  The constructor is passed only a RigidBodyTree and a control config filename, and the RBT was itself constructed [just a few lines before](https://github.com/RobotLocomotion/drake/blob/59a62d2dbe484b713c6e25561bdc3d647a4d17a7/drake/systems/controllers/constructQPDataPointerMex.cpp#L34).  

Although the tests depend on Gurobi, they fail before Gurobi is actually used.  Although the failure occurs inside a mex function, the interface of that function is just a bunch of filenames.  This gave me hope that I could reproduce the segfault without either MATLAB or Gurobi, simply by excising the behavior from line 34 of the mex to the point of segfault.  To that end, I hacked together a [test program](https://gist.github.com/david-german-tri/aceb8e45f6802d8db113baeb756b205b).  Unfortunately, it runs without segfault, and runs under valgrind with no reported violations.

@siyuanfeng-tri, @RussTedrake suggested I assign this to you for further investigation.

tkoolen, 2016-09-16:
It appears that changing `InstantaneousQPController::body_or_frame_name_to_id` from `unordered_map` to `map` and changing `computeBodyOrFrameNameToIdMap` to match fixes the segfault. I don't understand why. Since it's just with GCC and just in debug mode, it may be a compiler/std lib issue (note that the Mac/Release issues are resolved by https://github.com/RobotLocomotion/drake/pull/3462).

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri