RobotLocomotion/drake #3149: Certain long-running MATLAB tests sporadically time out in CI

david-german-tri, 2016-08-15:
* runPassiveDownhill
* testMultipleTaskSpaceRRTStar
* testTaskSpaceRRT

They are particularly timeout-prone on Mac in debug mode.  We should consider either extending the timeouts or finding ways to shorten the test.

ggould-tri, 2016-08-24:
This also appears to apply to: 
 * RigidBodyManipulatorMemoryTest
 * systems/plants/constraint/test/RelativeGazeTargetConstraintTest
 * systems/plants/constraint/test/RelativeQuatConstraintTest
 * systems/plants/constraint/test/testPostureConstraint

Because long-running matlab tests were just added to the continuous build, all of these tests will be disabled.  This issue will remain open, and the disabling will reference this issue in both source comments and commit messages.

jamiesnape, 2016-08-30:
Also:
* examples/Atlas/runDRCPracticeTerrain

RussTedrake, 2017-02-03:
i believe these have been disabled and will not be fixed (in the matlab versions)

closed 2017-02-03 by RussTedrake