RobotLocomotion/drake #3150: automation to prevent regressions of DOS-style line endings

rpoyner-tri, 2016-08-15:
A loose end from #3029: add/verify automation to guard against regression of DOS-style line endings.

jamiesnape, 2016-08-16:
Easiest thing would be to add it to cpplint_wrapper.py so it runs on the same build as cpplint.

mwoehlke-kitware, 2016-08-16:
Why not in a pre-commit hook? (I suppose we could do both for that matter, but it seems convenient to enforce it *before* CI, closer to the point the developer is likely to be thinking about it, and before it costs AWS $$.)

jamiesnape, 2016-08-16:
Do both.

jwnimmer-tri, 2016-08-17:
Given the paucity of folks who submit PRs from Windows, I wonder if a simple cpplint-like rule on PRs would be good enough?  Yes, it delays error reporting for folks who risk adding DOS files but don't run cpplint tests locally, but is that a problem we think is worth spending money on?

david-german-tri, 2016-08-18:
I agree with @jwnimmer-tri, catching this at lint is good enough.

mwoehlke-kitware, 2016-08-18:
TBH, It's not obvious to me how one goes about adding a not-cpplint check to `cpplint_wrapper.py`. There aren't any at present, and I'm not sure what's the correct way to do it so it shows up in CI reasonably.

By contrast, adding a hook is trivial, aside from the current hooks system being rather limited. (Fixing that would also make it trivial to add any other hooks we might want in the future.)

jamiesnape, 2016-08-18:
Add a python method that runs before or after the cpplint, ensure that the output includes one of the CTest warning regexes.

david-german-tri, 2017-01-13:
I think we could bolt Drake-specific linters into [cpplint.bzl](https://github.com/RobotLocomotion/drake/blob/master/tools/cpplint.bzl) easily enough, which would be sufficient to close this issue in my book.  Let me give it a try.

PR #4785 (2017-01-18): Add an infrastructure for Drake lint checks.
Fixes #3150.

@rpoyner-tri for feature review, @jwnimmer-tri for platform review

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/4785)
commit 9a1fe060 (2017-01-18): Add an infrastructure for Drake lint checks.

Fixes #3150.
closed 2017-01-18 by david-german-tri