RobotLocomotion/drake #3156: Problems during Source installation on Windows

chang-liang, 2016-08-16:
Never succeed in "Compile Drake" in "Drake Windows Installation Instructions"[http://drake.mit.edu/windows.html](url). 
After i typed in the codes bellows, (**July 29**)
`$ cd [build artifacts directory]
$ cmake --build . --target ALL_BUILD --config **RelWithDebInfo**`
i got this problem shown on Windows PowShell which you can see here [http://pan.baidu.com/s/1o8fFRTK](url)
And, 
after i typed in the codes bellows, (**Aug 16**)
`$ cd [build artifacts directory]
$ cmake --build . --target ALL_BUILD --config ** Release**`
i got this problem shown on Windows PowShell which you can see here [http://pan.baidu.com/s/1o8fFRTK](url)
And it stoped in Picture4 after Compiling for about 1hr.

Software version :

- Visual  Studio 14 2015 Win64

- Matlab 2016a

- Cmake 3.5.2

chang-liang, 2016-08-18:
stuck again in here today... please help... @jwnimmer-tri  , @hongkai-dai , @liangfok .
[image]
[image]
[image]
[image]
[image]
[image]

jwnimmer-tri, 2016-08-18:
I apologize, but I cannot offer windows help.  Perhaps someone from robot locomotion group can assist.

david-german-tri, 2016-10-06:
Windows is no longer supported.

closed 2016-10-06 by david-german-tri