RobotLocomotion/drake #3158: Implement helper methods to sanity check SystemOutput and Context.

amcastro-tri, 2016-08-16:
Implement methods `System<T>::IsOutputValid(SystemOutput<T>& output)` and `System<T>::IsContextValid(ContextBase<T>& context)` that check the consistency of systems' outputs and contexts when passed to methods `EvalOutput` and `EvalTimeDerivatives` since right now we are copying these code for each primitive.

david-german-tri, 2016-08-16:
:+1: Great idea.

sherm1, 2016-08-16:
I'm sad that both Outputs and Contexts have to be checked. Since the value for an output must reside in the Context, we should only have to verify that the Context is good. This is related to #3155.

jwnimmer-tri, 2016-08-16:
Consider that, instead of having a boolean "am I sane?" predicate that folks have to remember to check, a sugar API to obtain the handle you need ("the one true output port") embedded the check, so the only way you could (easily) get the handle was if the check passed.  Fewer keystrokes for users; more certainty for us that the checks are performed.

david-german-tri, 2016-08-17:
> Since the value for an output must reside in the Context...

This is not at all true right now.  

> we should only have to verify that the Context is good

Even if we did require that all outputs are references to a line of cache, I don't understand how this follows.  It would still be important to check that the output structure has the correct number of ports, if nothing else.

> Consider that, instead of having a boolean "am I sane?" predicate that folks have to remember to check, a sugar API to obtain the handle you need ("the one true output port") embedded the check, so the only way you could (easily) get the handle was if the check passed.

I agree this would be a good use of the predicates, but implementing them at all is a good start.

PR #3173 (2016-08-18): Helper methods to sanity check SystemOutput and Context.
Closes #3158.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3173)
closed 2016-08-18 by amcastro-tri