RobotLocomotion/drake #3167: Running more than one instance of Jenkins

jwnimmer-tri, 2016-08-17:
https://drake-jenkins.csail.mit.edu/ sayeth
```
Jenkins detected that you appear to be running more than one instance of Jenkins
that share the same home directory '/var/lib/jenkins'. This greatly confuses Jenkins
and you will likely experience strange behaviors, so please correct the situation.

This Jenkins:    327276788 contextPath="" at 6306@ip-172-31-29-90.ec2.internal
Other Jenkins:    1017653788 contextPath="" at 98119@ip-172-31-29-90.ec2.internal
```

jwnimmer-tri, 2016-08-17:
/CC @david-german-tri @jamiesnape @BetsyMcPhail

jamiesnape, 2016-08-17:
Where was this displayed?

jwnimmer-tri, 2016-08-17:
When I opened that top-level URI (or any other one), that message was displayed in lieu of the typical UI.  The only other element on the screen was a header bar with "enable / disable auto-refresh".  It also had a button like "Continue anyway" that I decided not to push.

jwnimmer-tri, 2016-08-17:
The sequence of events was something like:
- around 5pm(?) jenkins had like 50 builds in progress and 100 builds stalled and backed up
- around 7pm(?) jenkins looked like it reset, and was back to 0 builds; you could fire off new builds in the regular way (push to a PR, post a comment, etc.); some of these failed tests with "port already in use"
- around 9pm(?) jenkins started showing this message instead of the real UI
- this morning it was back to the regular UI

jamiesnape, 2016-08-17:
I am guessing some sort of JVM issue took Jenkins down uncleanly. I will have to go through the log files to be sure.

david-german-tri, 2016-08-18:
Since the outage is over, I'm downgrading this to `priority: medium`, but I would like to know the root cause at some point.

jamiesnape, 2016-08-18:
>  I would like to know the root cause at some point.

I will certainly try, but there is not much information to help diagnose.

david-german-tri, 2016-09-22:
Closing this as infeasible.

closed 2016-09-22 by david-german-tri