RobotLocomotion/drake #3170: Should we allow to flag input ports as direct feedthrough?

amcastro-tri, 2016-08-17:
With #3102 merged we could flag input ports as being direct feedthrough or not (by providing a bool member in `SystemPortDescriptor`).
Then the the method `System<T>::has_any_direct_feedthrough()` could have a default implementation that simply scans `System<T>::input_ports_`.
The user then would only needs to declare input ports as being direct feedthrough (default to `true`) or not  in their systems' constructors.

Would this be the right approach @david-german-tri and @sherm1?

david-german-tri, 2016-08-17:
We could do that, but eventually, the System needs to report the entire sparsity matrix.  I'm not sure reporting the logical-OR of that matrix across rows/columns is a useful intermediate step.

cc @bradking

amcastro-tri, 2016-08-17:
What are you calling the "sparsity matrix"? the sparsity on what?

david-german-tri, 2016-08-17:
For each output, does each input feed directly through to it.

bradking, 2016-08-17:
Once we get it working the functional form query can give us granular information all the way down to how each component of each output vector relates to each component of each input vector and state vector.

amcastro-tri, 2016-08-17:
What if (for instance) the `Adder` constructor would look something like:

```c++
OutputPort output(kVectorValued, length, kInheritedSampling);
InputPort input(kVectorValued, length, kInheritedSampling);
input.set_direct_feedthrough_on(output);
this->DeclareInputPort(input);
this->DeclareOutputPort(output);
```

I think that would contain all the information to build the sparsity matrix of inputs on outputs.

jwnimmer-tri, 2016-12-22:
Seems like this should be either assigned to @david-german-tri or closed?

sherm1, 2016-12-22:
We will be able to calculate the OutputPort-to-InputPort dependency matrix automatically when the cache is fully functional per #4364. We will have explicit dependencies for all cached quantities including OutputPorts, because those dependencies are needed for cache invalidation. (If InputPort i directly or indirectly invalidates OutputPort o then there is direct feedthrough from i to o.) The method proposed here won't be needed. I've added this to the things-to-do list in #4364 and will close this issue.

closed 2016-12-22 by sherm1