RobotLocomotion/drake #3171: Create more flexible git hooks system

mwoehlke-kitware, 2016-08-17:
For #3150, I want to add a new pre-commit hook.

The current system is a little clunky. On other projects, we have used a separate hooks repository and a "chain" hook that dispatches multiple hooks organized in directories.

I'd like to set up a similar system for Drake. Using a separate repository greatly eases maintenance; it is not necessary to muck about with hook version schemes, as we would just use the latest version of the hooks repository (the submodule would be tied to "master" rather than a particular SHA). The directory system makes it much easier to add and manage hooks.

This will of course necessitate creation of a hook repository, preferably in the RobotLocomotion organization.

I'm opening this issue initially for discussion. @bradking, @jamiesnape, @david-german-tri, @RussTedrake and anyone else are invited to weigh in.

bradking, 2016-08-17:
For reference, the current system has CMake install [this pre-commit hook](https://github.com/RobotLocomotion/drake/blob/0810fedc5548e21c0986bed34e2f24dbe40df949/cmake/git/pre-commit.in) in `.git/hooks`.  This is a minimal forwarding hook that should not need to change.  It reads [the .hooks-config file](https://github.com/RobotLocomotion/drake/blob/0810fedc5548e21c0986bed34e2f24dbe40df949/.hooks-config) to decide what hook to run from within the source tree.  Currently it runs [this pre-commit.bash script](https://github.com/RobotLocomotion/drake/blob/0810fedc5548e21c0986bed34e2f24dbe40df949/cmake/git/pre-commit.bash) for the actual checks.

In order to chain to multiple hooks we would simply change `pre-commit.bash` to load hooks from a directory.

mwoehlke-kitware, 2016-08-17:
So, @bradking and I had a fairly long offline discussion here, that isn't immediately relevant. In the short term, however, what I'm looking for is a consensus whether or not it makes sense to have the hooks tracked in a separate repository, the idea being that developers will always have the latest hooks even on an old branch. (Note: my current design will *also* allow for hooks tied to the drake version, if we should decide we need those. That is, you can have a mix of both; some hooks that are version-tied, and some that aren't.)

I propose RobotLocomotion/git-hooks for said repository.

mwoehlke-kitware, 2016-08-18:
Can someone with permission create the repo and give me write access?

david-german-tri, 2016-08-19:
https://github.com/RobotLocomotion/drake-git-hooks

david-german-tri, 2017-04-18:
I'm dropping the priority to "backlog" - this could be nice, but there's little demand for it.

jwnimmer-tri, 2017-06-19:
Given that we're going to be _removing_ the one existing pre-commit hook, I'm not sure that generalizing the hook system would be useful.  Should we just close this?

jamiesnape, 2017-06-19:
We potentially may want a hook to help with #6124, but I cannot think of other instances, so I would have no objections.

jwnimmer-tri, 2018-11-14:
Not relevant anymore.

closed 2018-11-14 by jwnimmer-tri