RobotLocomotion/drake #3187: Audit superbuild to assess where downloads are coming from

jwnimmer-tri, 2016-08-18:
In #2771, we were downloading a tarball from Zurich, on an old FTP server that was slow and eventually failed, which meant all of our builds broke.  That's not quite to the level of leftpad, but we can do better.

With all externals turned on, we should see where any "download from the internet" are happening, and ensure we have an MIT- or TRI-controlled distribution mechanism -- either as a backup as least, or as the canonical location if possible.

I assert that Drake should download exclusively from github or Amazon, except where licensing concerns prevent us from redistributing the artifacts.

If possible, it would be awesome if CI could enforce this by firewalling the general internet, and then only whitelisting certain servers.

jamiesnape, 2016-08-19:
The firewalling is a little tricky as it would have to be IP address based and the relevant IP ranges are not guaranteed.

jamiesnape, 2016-08-19:
Or at least I would have to install and configure and completely different firewall or a proxy to be able to do so.

david-german-tri, 2016-08-19:
I'm kind of surprised by that, and would like to know more, but let's take this conversation off public GitHub.

jamiesnape, 2016-08-19:
Sure.

jamiesnape, 2016-08-19:
Eigen comes from bitbucket. I would have thought that would be fine, but I can put a tarball in S3 if you prefer.

jamiesnape, 2016-08-19:
GTK+ comes from win32builder.gnome.org.

jwnimmer-tri, 2016-08-19:
I have a final answer or strong opinion about which servers we should rely on.  Perhaps we can collect the list of non-Amazon, non-github servers, and then take an overall decision?

liangfok, 2016-08-19:
^^^ Missing "don't after "I"?

david-german-tri, 2017-04-11:
This is trivial in the Bazel WORKSPACE file, so I'm going to close as wontfix for CMake.

closed 2017-04-11 by david-german-tri