RobotLocomotion/drake #3199: cpplint_wrapper_test fails on Windows

sherm1, 2016-08-19:
Starting with a fresh clone of drake I did a virgin Debug build on Windows (64 bit). Superbuild was clean in Visual Studio. I switched to build/drake/drake.sln; build there was clean (and fast since everything was already built). I ran the test suite and everything worked except `cpplint_wrapper_test`. 

I do have cpplint in my path and it appears that the test ran but failed. I didn't see it in the list of tests or in the bin directory so I'm not sure how to get more information about the failure.

Some not-very-helpful output:
```
1>        Start 14: cpplint_wrapper_test
1>  14/87 Test #14: cpplint_wrapper_test .....................***Failed    4.02 sec
The following tests FAILED:
1>  	 14 - cpplint_wrapper_test (Failed)
1>  Errors while running CTest
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.CppCommon.targets(133,5):
error MSB3073: The command "setlocal ... VCEnd" exited with code 8.
```

jwnimmer-tri, 2016-08-19:
Well, I wrote the test, but I can't puzzle out what's failing.  Is that a way on windows to ask for more verbose output?

sherm1, 2016-08-19:
> Is that a way on windows to ask for more verbose output?

I can get more out of it if I can figure out how to run it directly from a shell -- is there an executable somewhere?

liangfok, 2016-08-19:
The executable is the python interpreter itself.

```
python drake-distro/drake/common/test/cpplint_wrapper_test.py
```

sherm1, 2016-08-19:
Oh -- thanks, @liangfok!

sherm1, 2016-08-19:
Below is actual output. Does this speak to you at all @jwnimmer-tri?
```
C:\Users\Sherm\Documents\GitHub\sherm1>python drake-distro/drake/common/test/cpplint_wrapper_test.py
.FF.FF
======================================================================
FAIL: test_huge (__main__.TestStringMethods)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "drake-distro/drake/common/test/cpplint_wrapper_test.py", line 104, in test_huge
    r"TOTAL 25 files passed"])
  File "drake-distro/drake/common/test/cpplint_wrapper_test.py", line 37, in run_and_expect
    self.assertRegexpMatches(output, item)
AssertionError: Regexp didn't match: 'Checking \\.+\\n' not found in 'Checking ............................\r\n TOTAL 25 files passed\r\n'

======================================================================
FAIL: test_ignored_extension (__main__.TestStringMethods)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "drake-distro/drake/common/test/cpplint_wrapper_test.py", line 82, in test_ignored_extension
    r"TOTAL 0 files"])
  File "drake-distro/drake/common/test/cpplint_wrapper_test.py", line 37, in run_and_expect
    self.assertRegexpMatches(output, item)
AssertionError: Regexp didn't match: 'Ignoring .*/cpplint_wrapper_test.py' not found in 'Ignoring C:\\Users\\Sherm\\Documents\\GitHub\\sherm1\\drake-distro\\drake\\common\\test\\cpplint_wrapper_test.py; not a valid file name.\r\nChecking ...\r\n TOTAL 0 files passed\r\n'

======================================================================
FAIL: test_summarize (__main__.TestStringMethods)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "drake-distro/drake/common/test/cpplint_wrapper_test.py", line 51, in test_summarize
    r"TOTAL 1 files passed"])
  File "drake-distro/drake/common/test/cpplint_wrapper_test.py", line 37, in run_and_expect
    self.assertRegexpMatches(output, item)
AssertionError: Regexp didn't match: 'Checking \\.+\\n' not found in 'Checking ....\r\n TOTAL 1 files passed\r\n'

======================================================================
FAIL: test_true_positive (__main__.TestStringMethods)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "drake-distro/drake/common/test/cpplint_wrapper_test.py", line 67, in test_true_positive
    r"TOTAL 1 files passed"])
  File "drake-distro/drake/common/test/cpplint_wrapper_test.py", line 37, in run_and_expect
    self.assertRegexpMatches(output, item)
AssertionError: Regexp didn't match: 'Checking \\.+\\n' not found in 'Checking ....\r\n TOTAL 1 files passed\r\n'

----------------------------------------------------------------------
Ran 6 tests in 3.065s

FAILED (failures=4)
```

sherm1, 2016-08-19:
P.S. That is Python 2.7.10.

jwnimmer-tri, 2016-08-19:
```
AssertionError: Regexp didn't match:
  'Checking \\.+\\n'
not found in
  'Checking ............................\r\n TOTAL 25 files passed\r\n'
```

DOS newlines for the win!  I wonder why CI isn't complaining.

jwnimmer-tri, 2016-08-19:
>  I wonder why CI isn't complaining.

The test is effectively disabled on win32 CI, because of this line that turns of google-styleguide:
https://github.com/RobotLocomotion/drake-ci/blob/1fc8b6138c476c6e96c5f79351155f444d8c73c9/ctest_driver_script.cmake#L760

jamiesnape, 2016-08-19:
There is no Python on the Windows CI images.

sherm1, 2016-08-19:
> DOS newlines for the win! I wonder why CI isn't complaining.

I'm not sure that means there are DOS newlines checked in. Git ensures that anything in the repo uses Linux-style line endings, but on Windows the workspace files (if that is the right term for the ones you see and work on) get Windows style endings locally.

jwnimmer-tri, 2016-08-19:
I just mean that the line separators emitted by cpplint when printing warnings are using `\r\n`, but my unit test (where cpplint itself is the DUT) doesn't allow for `\r` at all.  If you want to edit the `cpplint_wrapper_test.py` to admit an optional `\r` and PR that, I will review it for you.

PR #3201 (2016-08-22): Modify `cpplint_wrapper_test.py` to work on Windows
`cpplint_wrapper_test.py` had some Linux-specific comparisons that failed on Windows. This PR makes it accept either Linux or Windows line endings and path separators. It does not attempt to determine which platform it is running on -- it just makes the tests more forgiving.

Resolves #3199.

+@jwnimmer-tri for feature & platform review.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/3201)
commit 27f0dd42 (2016-08-22): Modify cpplint_wrapper_test.py to work on Windows.
Resolves #3199.
closed 2016-08-22 by jwnimmer-tri