RobotLocomotion/drake #3211: Update to Eigen 3.3 beta 2

amcastro-tri, 2016-08-22:
Thank you @tkoolen for noticing this. This would fix the issue with #3145

tkoolen, 2016-08-22:
Just self-assigned, let me know if you're already on it.

amcastro-tri, 2016-08-22:
Great! thank you! I'm glad you want to do this but please don't feel obliged to.

tkoolen, 2016-08-22:
I'm running into compilation issues within `Eigen::PolynomialSolver` after updating. Trying to figure it out.

amcastro-tri, 2016-08-22:
I figured it wouldn't be that easy, take your time. I actually still didn't close the bug issue on Eigen just to see what feedback we get from them.

tkoolen, 2016-08-22:
Just going to keep commenting on problems I'm finding and solutions for future reference.
The `PolynomialSolver` problem goes away after updating AutoDiffScalar.h to the version in master: https://bitbucket.org/eigen/eigen/src/0b62bc896f8070c48214ffc447bc597c261933bc/unsupported/Eigen/src/AutoDiff/AutoDiffScalar.h?fileviewer=file-view-default

Not there yet.

tkoolen, 2016-08-22:
Please do close that bug by the way. We know what happened, it's fixed in master, and you'd be wasting their time otherwise.

tkoolen, 2016-08-22:
Just copying over the AutoDiffScalar file from master causes other problems unfortunately...

tkoolen, 2016-08-22:
I'm having much better luck with the latest default branch (unmodified, hg 9431:5a47e5a5b02e). There's only one issue with `operator/` for `drake::functional_form` (used as custom scalar type for Eigen matrices).

jwnimmer-tri, 2016-08-23:
Github comment parsing was overly-optimistic.  Drake still needs to bump its `eigen-pod` version.

amcastro-tri, 2016-08-23:
@tkoolen, the Eigen people asked me if migrating to 3.3 actually does solve our issue. See here http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1279

Is that correct? it seems like migrating to 3.3 causes even more issues?

tkoolen, 2016-08-24:
3.3 beta 2 fixes the issue with #3145, but introduced several issues related to AutoDiffScalar. The fixes for those haven't made it into any released version yet.

amcastro-tri, 2016-08-24:
So #3227 fixes this? and should we document the 
>  introduced several issues related to AutoDiffScalar

?

tkoolen, 2016-08-24:
Yes.

Bugs fixed since 3.3 beta 2:
* http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1260
* http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1261
* http://eigen.tuxfamily.org/bz/show_bug.cgi?id=1264

closed 2016-08-24 by tkoolen