RobotLocomotion/drake #3223: Bad attributes in mexfiles break mac matlab builds

ggould-tri, 2016-08-23:
#3605 appears to have caused a nightly CI failure at https://drake-cdash.csail.mit.edu/viewBuildError.php?buildid=88374 .  This is acknowledged at #3206 in the comment at https://github.com/RobotLocomotion/drake/pull/3206#issuecomment-241474837 .

The failure text is a number of instances of:
```
/Users/elcapitan/workspace/mac-clang-nightly-matlab-debug/drake/util/drakeMexUtil.cpp:457:10: error: an attribute list cannot appear here
template DLLEXPORT mxArray* eigenToMatlabSparse(MatrixBase<MatrixXd> const&,
         ^~~~~~~~~
/Users/elcapitan/workspace/mac-clang-nightly-matlab-debug/drake/../drake/util/drakeMexUtil.h:38:19: note: expanded from macro 'DLLEXPORT'
#define DLLEXPORT [[gnu::visibility("default")]]
```

Assigning to @mwoehlke-kitware as he is responsible for #3605 and #3206.

Because #3206 indicates that this is a known issue being actively worked, I am not dropping the revert hammer on it just yet.

jwnimmer-tri, 2016-08-23:
Was #3210 the fix for these, perhaps?

mwoehlke-kitware, 2016-08-23:
> Was #3210 the fix for these, perhaps?

Yes. (Supposed to be at least... is it still failing? That *just* got merged, so the failures may predate it...)

mwoehlke-kitware, 2016-08-23:
(Oh, I see... yes, #3210 didn't land in time to fix last night's nightlies, and there are no corresponding continuous builds that show this error.)

ggould-tri, 2016-08-24:
I confirm that the build failure is no longer present in nightly (that nightly is yellow, but for other reasons) and so am closing this issue.

Thank you.

closed 2016-08-24 by ggould-tri