RobotLocomotion/drake #3224: testGeometryGradientsComparison fails in nightly

ggould-tri, 2016-08-23:
The test "testGeometryGradientsComparison" fails in the linux-gcc-nightly-matlab-debug build.  In fact it has been failing for as long as that build has existed (August 8, 2016) and may have been present for quite some time.

Representative failure: https://drake-cdash.csail.mit.edu/testDetails.php?test=1919619&build=86840
First known failure: https://drake-cdash.csail.mit.edu/viewTest.php?onlyfailed&buildid=73533

Because this is a matlab failure I am tagging it to RLG to bisect.  If it turns out to be a result of mex calls into the new solver stack, it should be reassigned to the manipulation team.

jwnimmer-tri, 2016-08-23:
Per recent discussions, tests that are known to fail under certain conditions should not be left to linger and pollute the nightly build output.  If it isn't fixed this week, we should disable the test in the `CMakeLists.txt` for the settings that are known to fail (leaving this issue open to track the fix).

ggould-tri, 2016-08-24:
Per @jwnimmer-tri above, #3239 does just that.

jwnimmer-tri, 2016-08-25:
@ggould-tri My proposal was to disable in `Debug` mode only, but continue to run it in `NOT Debug` modes such as `Release.  (Assuming it was only failing in Debug.)

ggould-tri, 2016-08-25:
Hm, I'm a bit unclear on whether the failure is debug-only.

Given that this is a correctness error rather than a timeout, it would be
quite curious if it only failed in debug and that suggests possibly faulty
test design (eg, assertions that are not turned on in release).

I will investigate this later today.

ggould-tri, 2016-09-06:
Okay, the extremely unlikely appears to be true -- this test was running on both debug and release but only failing in debug.  The matlab code in question does not appear to perform the check in only one case.  The assertion is matlab and not in mexland.

Thus the underlying issue must be that the mex code actually is producing different values in the debug and release cases.

Because mex is a doomed backwater I am not going to flag this up to too high a priority.  If RLG gets into their backlog, though, this has a nasty code smell, so I'm declaring priority medium pending our next bug triage.

jwnimmer-tri, 2017-06-19:
Closing as won't-fix, because #6369 will be removing the relevant software.

If I'm confused and this should in fact stay open, please re-open an clarify the remaining unsolved problem(s) in a new comment here.

closed 2017-06-19 by jwnimmer-tri